kr15h/ofxPiMapper #29: After loading the app on RPi, switch video sources then app crashes and this is the output:(3 case)

lanli, 2015-09-08:
case1:                    
[notice ] MediaServer: Initialized reference count of /home/pi/of_v0.8.4_linuxarm7l_release/addon/ofxPiMapper/example/bin/data/sources/videos/test.mov to 1
Segmentation fault
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:168: recipe for target 'run' failed
make: ***[run]Error 139

case2:(sometime) 
[warning] bool ofxOMXPlayerEngine::setup(ofxOMXPlayerSettings&): FAST PATH MOVE OPEN FAILED - LIKELY A STREAM, TRYING SLOW PATH
[ error ] COMXPlayer::OpenFile - xxx......

case3:(enableAudio=true)
[ error ] COMXAudioCodecOMX::GetChannelMap - FFmpeg reported 2 channels, but the layout contains 0 ignoring
...

kr15h, 2015-09-08:
Thanks for the feedback @lanli. Will look into that as soon as possible.

kr15h, 2016-06-02:
Seems to be fixed now.

closed 2016-06-02 by kr15h