kr15h/ofxPiMapper #31: make: *** [run] Error 139 or 135

luisa20, 2015-10-20:
I have two errors when I use an own new video source. What can I do?

kr15h, 2015-10-20:
You should try re-encoding it. The following are the ecoding recomendations from ofxOMXPlayer:

 - Use MPEGStreamclip (http://www.squared5.com/)
 - H.264
 - Quality 50%
 - PCM Audio (more compatible with HDMI)

luisa20, 2015-10-20:
Thanks!
I have try re-encoding. It did not work.
I copied the video in the folder: /home/pi/openFrameworks/addons/ofxPiMapper/example/bin/data/source/videos

kr15h, 2015-10-21:
What is the full name of the video and would it be possible for you to send it to me to take a look?

luisa20, 2015-10-21:
I have the video re-encoding again:
h.264 1280x720
24 fps
qualität 50%

now, it work!

Thanks for help!

kr15h, 2015-10-21:
Great! Thanks for letting me know!

closed 2015-10-21 by kr15h