kr15h/ofxPiMapper #32: Can I make a autorun and when yes how to make it?

luisa20, 2015-10-22:


newemka, 2015-10-22:
I've managed to make it but I don't remembrer how. Here is a link that should help you. I think that's how I did it. https://neverbenever.wordpress.com/2015/02/11/how-to-autostart-a-program-in-raspberry-pi-or-linux/

kr15h, 2015-10-22:
Try this: 
 - http://kr15h.github.io/RPi-Setup/index.html#titles.5.5
 - http://kr15h.github.io/RPi-Setup/index.html#titles.5.6
 - http://kr15h.github.io/RPi-Setup/index.html#titles.5.7

And add the full path to the executable e.g. `/home/pi/openFrameworks/addons/ofxPiMapper/example/bin/example -f`

`-f` for fullscreen

luisa20, 2015-10-22:
Thank you for reply! 
It don't work :(
below 5.5
Preventing the screen from going blank #
Add the following line just before the exit 0 line:
sh -c 'setterm -blank 0 -powersave off -powerdown 0 < /dev/console > /dev/console 2>&1'

setterm? is this correct

I have an error Message: 
/etc/rc.local: Syntax error: Unterminated quoted string [fail] startpar: service(s) returned failure: rc.local ... failed!

kr15h, 2015-10-23:
You have to follow the instructions carefully starting from this section: http://kr15h.github.io/RPi-Setup/index.html#titles.5.5
Make sure you open the correct files and add the lines mentioned in the text.

kr15h, 2015-10-23:
And regarding the error - check whether you have the closing quote ('). Its the syntax error. Probably you accidentally did not select the end quote. It has to be there. The whole line. With qotes.
```
sh -c 'setterm -blank 0 -powersave off -powerdown 0 < /dev/console > /dev/console 2>&1'
```

luisa20, 2015-10-23:
Yes I have the (') forget. Now, I have it includet and then ctrl x, y,
enter, sudo reboot enter, but it don't work. Must I make  5.5/5.6/5.7 again?

kr15h, 2015-10-24:
Could you specify what is the full path to your openFrameworks directory and are you using the add-on example or your own app?

luisa20, 2015-10-24:
so, I have make all completely new and now it works !!!! :)

Thanks for help!

Now, I will make a FBO sources ...

kr15h, 2015-10-24:
Yes, that's where the fun begins. Make sure that you pull in the latest changes from the `dev` branch - I discovered some bugs related to FBO sources, they are fixed now, but only in the `dev` branch. To do that, go to the ofxPiMapper addon directory
```
cd /home/pi/openFrameworks/addons/ofxPiMapper
```
Switch to the `dev` branch
```
git checkout dev
```
And pull in latest changes from remote `dev` branch
```
git pull origin dev
```
And now make that FBO source!

closed 2015-10-24 by kr15h

luisa20, 2015-11-14:
Hi,
and what mean that?:

ofxPiMapper * mapper = new ofxPiMapper();
CustomSource * source = new CustomSource();
mapper.addFboSource(*source);
mapper.setup(); // Setup must be run after adding the FBO source

Thanks for your help!!!