kr15h/ofxPiMapper #34: Problems with Raspbian Jessie

mojibaked, 2015-12-05:
I couldn't get the example to work with Jessie.
After reading in the chat i tried installing Raspbian Wheezy, which did work.

kr15h, 2015-12-05:
I believe I have to do serious testing weekend with all of those.

mcanet, 2015-12-06:
any plans to adapted to OF0.9? that works in Jessie

stephanebordonaro, 2015-12-20:
hello kr15h,
the example works for me on jessy /pi2 .
i am able to add surface and delete some ...but impossible to add texture ( image or video ) .
on menu " 3" i select the surface then on menu"'4" the images and vidéos are displayed on the left corner but impossible to clic to select any .( so the square are on "damier" type ) ...
do you any clues ? 

thx a lot

kr15h, 2015-12-20:
It is possible that there is not enough memory allocated to the GPU. Enter the Raspberry Pi configuration utility.
```
sudo raspi-config
```

And under the `Andvanced` menu item there should be the `Memory Split` option. Select that and allocate at least 128 MB to the GPU.

Hope this helps. I have to try out the latest Pi and Raspbian myself, this is my first holiday, will do it the following week. So excited... ; D

stephanebordonaro, 2015-12-20:
hello,
i have checked the memory split ...i have set it to 256 ( before the ) ...doesn't work ...
moreover ...is there a way to start the prog in full screen ? and to start it directly during the boot ? 

finally ; is there a possibility to do a workshop in france ? i think that it will really interest the french raspberry community !

thx for your reply ...and have good holiday !

kr15h, 2015-12-21:
OK, I have to check this out, have to do a full install cycle with Jessie. As for now my suggestion would be to try Wheezy as that works. 

Workshop in France would be cool to do. It's just a matter of planing, networking. I mean, if you or someone you know would be into finding place and time for that. Some funding, at least for the travel expenses, would be cool. And if someone there takes care of the propaganda aspects, getting the right amount of participants for that.

stephanebordonaro, 2015-12-23:
hello ,
question concerning a workshop in france ...what would be the ideal amount of participant ? how many days of work  would be interesting ? Do you think a one or one half day would be interesting integrated in a 4/5 days more large workshop about mapping ( mad mapper ,..)?

kr15h, 2015-12-23:
It depends on the level of the participants. Usually minimum 2 days. 

 - Day 1: openFrameworks basics, installing addons, using ofxPiMapper example, launching on boot
 - Day 2: Creating custom generative source with ofxPiMapper

More days for more fun of course, but it can get tricky. 10 participants would be OK. Some programming experience would be great, especcially for day 2.

mojibaked, 2015-12-24:
The code is great by the way, works like a charm on Raspian Wheezy!
I'm rendering a live 720p rtp video stream onto a quad mapped surface, and getting 57-60 fps. We're now using 3 Raspberry's in sync.
Raspberry PI 2 cpu load is around 10% while running the application. Really nice!
Still working on a bug which causes the application to crash with segmentation faults, this happens when the current video ends, and server sends out the next video in the playlist. I'm guessing i have to make ofxOMXPlayer listen for connections, instead of trying to read frames that don't exist.

kr15h, 2015-12-24:
Thanks for your feedback @mojibaked !

kr15h, 2015-12-25:
Just tested with Raspbian Jessie lite, 256 MB memory split, example works, example videos play smoothly.

stephanebordonaro, 2015-12-25:
thx kr15h to have a made correction to your code !
to mojibaked : hello , could you give me some clues to integrate sync between some raspberry ...maybe inside ofxpimapper?

kr15h, 2015-12-26:
There are still problems with source selection in oF v0.9.0. I'm on it.

commit b33b576e (2015-12-25): Update makefile for of 0.9.0, close #34
commit 25770a92 (2015-12-26): Fix source selection for oF v0.9.0 and above, close #34
closed 2015-12-26 by kr15h