kr15h/ofxPiMapper #35: is there a way to boot at startup  directly to ofxpimapper in full screen ? (noobs question ...)

stephanebordonaro, 2015-12-20:


kr15h, 2015-12-20:
It seems that I have to add instructions on how to do this, but for now read the following.
`http://kr15h.github.io/RPi-Setup/index.html#titles.5.5`

And then in the `startup.sh` add a line that points to the compiled binary of your application (or ofxPiMapper example application).

**startup.sh**
```bash
#!/bin/bash

echo "Checking SSH connection...";
if [ -z "$SSH_CONNECTION" ]; then
	echo "No SSH connection, launching scripts.";
	/home/pi/openFrameworks/addons/ofxPiMapper/example/bin/example
else
	echo "SSH connection, nothing to be launched.";
fi
```
Before doing this it is essential that you run `make && make run` for your app.
```bash
cd /home/pi/openFrameworks/addons/ofxPiMapper/example
make && make run
```

Let me know if you have problems, will update this.

closed 2015-12-20 by kr15h

stephanebordonaro, 2015-12-21:
Hi cool works great ! Thx 

Stéphane Bordonaro 
Créateur vidéo et lumière 
stephane.bordonaro@gmail.com
www.les4elements.com

-----Message d'origine-----

http://kr15h.github.io/RPi-Setup/index.html#titles.5.5

stephanebordonaro, 2015-12-21:
Hi ...thx for all This useful information.....i have also try to make it boot directly in full screen mode ... Changing things in the main prg is the right way ? I have try to force full screen ...without success ...

Stéphane Bordonaro 
Créateur vidéo et lumière 
stephane.bordonaro@gmail.com
www.les4elements.com

-----Message d'origine-----

http://kr15h.github.io/RPi-Setup/index.html#titles.5.5

kr15h, 2015-12-22:
You just pass the -f flag to the executable. I added faq with instructions about this today. https://github.com/kr15h/ofxPiMapper/blob/master/FAQ.md

stephanebordonaro, 2015-12-22:
yeah ...things coming more clear to a " one week " raspberry user ...
i have seen things to hide the boot loader and replace the raspberry logo
by yours or any ...

thx for your work

kr15h, 2015-12-22:
I think that to replace the Raspberry logo you have to recompile the kernel. At least that's the latest information I found when I was thinking about the same.