kr15h/ofxPiMapper #37: can't build under OSX

martialgallorini, 2016-01-15:
hello,

thanks for the addon
i have a problem : i can't make it to build under OSX El Capitan and oF09

see attached image 
[image]

any clue ?

kr15h, 2016-01-15:
It is possible that you are missing dependencies. Did you install ofxUI?

martialgallorini, 2016-01-15:
no ! i thought it was now using ofxGui ?

i will ha a try

kr15h, 2016-01-15:
I think that even though ofxOMXPlayer is not used on Mac, you should download that as well. I will adjust the code so that is not the case. This is a good point.

martialgallorini, 2016-01-15:
No chance using ofxUI
Also yes maybe a problem dealing with ofxOMXPlayer. I already had it installed though

kr15h, 2016-01-15:
I will take a look a bit later today. It might be that there have to be some adjustments in the xcode project settings.

martialgallorini, 2016-01-15:
ok thanks a lot.

here a screenshot in case it is of any help

[image]

kr15h, 2016-01-15:
Yeah it seems that it is trying to compile ofxOMXPlayer although xcode should not do it.

kr15h, 2016-01-15:
Maybe you should remove ofxOMXPlayer from the xcode project tree. See what happens. Is it actually there?

kr15h, 2016-01-15:
Another question - are you trying to build the example xcode project or are you building your own app?

martialgallorini, 2016-01-16:
I will try with removing completely ofxOMXplayer.
i am trying to bbuild the exemple code

kr15h, 2016-01-16:
yeah - the problem seems to be that on mac the regular openFrameworks video class is used. ofxOMXPlayer is used only when you compile the code on a Raspberry Pi.

kr15h, 2016-01-16:
Please let me know if removing ofxOMXPlayer did work.

martialgallorini, 2016-01-16:
ok, now it builds when ofxOMXPlayer is removed
but it doesn't seem to work properly : different modes seem to be the same

here are some logs if it helps

[image]

kr15h, 2016-01-16:
So it compiles and runs. It seems that there is a file missing. What kind of sources are you trying to load?

martialgallorini, 2016-01-16:
none ! just running the example

kr15h, 2016-01-16:
Do you have anything in the `bin/data` folder of the project?

martialgallorini, 2016-01-16:
yes, everything is in its right place...

martialgallorini, 2016-01-16:
i cleaned the build and re-built it entirely. Now it seems to be better.
but always those errors in log
[image]

kr15h, 2016-01-16:
Don't worry about the errors. They are more as warnings. I probably should remove them to not cause confusion like this. It's just that loading the video takes some time and before it is loaded, the rendering loop is running. If the texture object to be rendered is empty (video not loaded), ofxPiMapper logs this error.

closed 2016-01-16 by kr15h

martialgallorini, 2016-01-16:
perfect. Thanks a lot for your help and all your great work !