kr15h/ofxPiMapper #38: input source not accurate when quad perspective activated

martialgallorini, 2016-01-20:
For the record : 

once quad perspective is activated, selecting an area in the input source view results in non consistent output : output doesn't reflect the area selected

kr15h, 2016-01-20:
Yes, that's how the perspective warping works in a way. Try to select the whole texture as input. I will continue searching for alternatives and more precise solutions though. Can you post a screenshot here? I will open an issue with this.

kr15h, 2016-01-20:
See #39

martialgallorini, 2016-01-20:
great. Works perfectly now with homography implementation.
thanks a lot !

kr15h, 2016-01-21:
Someone had to do it! : ]

closed 2016-01-21 by kr15h

martialgallorini, 2016-01-21:
i would have if only i had the skills. but maths and i never happened to be the best friends back at school !