kr15h/ofxPiMapper #39: Once quad perspective is activated, selecting an area in the input source view results in non consistent output: output doesn't reflect the area selected

kr15h, 2016-01-20:
In a way this is because of how perspective warping works. @aspeteRakete used a method described here http://www.reedbeta.com/blog/2012/05/26/quadrilateral-interpolation-part-1/

I wonder what MadMapper perspective warping looks like, will check it out. In my opinion, the perfect solution is a combination of mesh warping and perspective warping.

martialgallorini, 2016-01-20:
here is some screenshots : 

Window size : 1280 x 720
Input size : 1280 x 720

Without perspective 
[image]

With perspective activated
[image]

martialgallorini, 2016-01-20:
Did you have a look at the ofxMtlMapping2D project ? It doesn't seem updated anymore but perspective warping seems to work there and maybe you could find some inspirations for perspective and/or other things

kr15h, 2016-01-20:
Good idea. Will check that.

martialgallorini, 2016-01-20:
ofxMtlMapping2D doesn't seem updated anymore but still builds in oF09 but crashes at some times. Interesting project though. I tried the perspective warping and it works great.

There is a dev branch which seems more advanced and more recent, but i couldn't make it build.

kr15h, 2016-01-20:
it is using the `homography.h` by Arturo Castro. I was initially looking at this as a solution. Behold.

martialgallorini, 2016-01-20:
yes, every project i had a test with uses openCV for the perspective correction or the homography.h from Arturo Castro as alternative. Arturo found this solution to prevent adding the whole openCV library just for the use of the perspective correction feature.

kr15h, 2016-01-20:
Added homography helper. Works!

closed 2016-01-20 by kr15h