kr15h/ofxPiMapper #42: Gstreamer problem

martialgallorini, 2016-02-11:
Hello,

I am facing a problem with latest commit of ofxPiMapper and oF0.9.2 under Ubuntu Trusty 64.

I installed oF and launched install_dependencies.sh and install_codecs.sh

Compiling went well but when running i get the messages : 

[error] ofGstUtils: embedded video playback halted for plugin, module source reported: Ressource not found
[error] ofGstUtils: startPipelin(): unable to pause pipeline
[error] ofGstUtils: embedded video playback halted for plugin, module decodebin1 reported: Your GSTreamer installation is mussing a plug-in

App is running. Images are ok, but video stays black

thanks

kr15h, 2016-02-11:
Have not tried with of 0.9.2 yet. Did you change the GPU memory setting in `raspi-config` to something above 128?

martialgallorini, 2016-02-11:
i am not using a RPi but an Intel Compute Stick

kr15h, 2016-02-11:
OK, that will be a bit hard to debug as I am not targeting the platform at all, but it would be interesting to try nevertheless.

kr15h, 2016-02-11:
I could try to see what could be wrong with of0.9.2

martialgallorini, 2016-02-11:
i am looking at which gstreamer package to install to get this decodebin1 plug in. Do you have a clue?

kr15h, 2016-02-11:
No idea : [

martialgallorini, 2016-02-11:
ok i re-installed every dependencies in install_dependencies.sh but i keep having errors.
it doesn't on a trusty VM on my mac

i have hard time build though. Kind of random errors

kr15h, 2016-02-12:
Yeah, this I can not solve at the moment as I don't have the Compute Stick.

kr15h, 2016-02-12:
Just tested with 0.9.2, everything seems to work. So it's something Compute Stick specific

martialgallorini, 2016-02-14:
I re-installed everything from OS to oF and addons
Seems to work now
probably something was broken !

thanks a lot

kr15h, 2016-02-14:
Then basically we can say that ofxPiMapper runs on Intel Compute Stick. That's awesome. Thanks for your input!

closed 2016-02-14 by kr15h

martialgallorini, 2016-02-14:
looking for some workarounds for some strange behavior though : 
- can't make it to work on headless system, but probably something dealing with Xorg or something
- fullscreen mode with -f option keeps showing window. See #43 

Even if it would be great to see some performance improvements and optimization on the video player (like AVFoundation player in MAC or OMXPlayer on RPi) it runs pretty well.

Using ubuntu 14.04 with openbox and one video at resolution 1280x720. Could be better, but smooth enough for my purpose.

thanks