kr15h/ofxPiMapper #43: fullscreen mode not fullscreen

martialgallorini, 2016-02-12:
hello,

when using fullscreen=true at app launch, app stays in window mode when starting.

using "f" key to make it fullscreen doesn't work the first time. Needs to use "f" twice to make it fullscreen.

i am under ubuntu 14.04 64 bits and oF0.9.2

kr15h, 2016-02-12:
You can also use the -f flag when launching the binary

martialgallorini, 2016-02-12:
yes i know. But same problem.

strange : it works on stock ubuntu (on a VM) but i have stripped down ubuntu version installed on a machine on which i switched Unity to LXDE, and here it doesn't work...

kr15h, 2016-02-12:
Will check this

martialgallorini, 2016-02-17:
hello,
any updates on this ?
thank you

kr15h, 2016-02-17:
Not yet

martialgallorini, 2016-02-17:
Same on debian Jessie.
Tried under openbox, XFCE and Gnome Shell : same problem. 
So it doesn't seem related to the Window Manager

kr15h, 2016-02-17:
Are you in desktop mode?

martialgallorini, 2016-02-17:
what do you mean by desktop mode ?

kr15h, 2016-02-17:
It seems more like a general problem with openFrameworks and linux window frameworks. I develop on mac and adjust to raspberry pi while using the command line.

kr15h, 2016-02-17:
With desktop mode I mean, does the desktop of yours have nice icons? Does a window open when you click on them? Or you launch the app from cli mode?

martialgallorini, 2016-02-17:
well yes i am in desktop mode. I tried to launch the app from command line (on a terminal window) or by clicking the icon. The app launches without problem.

I actually installed a fresh ubuntu and will give it a try. Then i will strip it down to a lightweight desktop environnement to see the difference. I will let you know.

thanks

kr15h, 2016-02-17:
Well yes, I kind of do not want to spend time targeting all environments and fixing bugs because of bugs elsewhere. As I said, I am developing on Mac and then making it work on the Raspberry Pi CLI only.

martialgallorini, 2016-02-17:
yes no problem, i really understand. you really can't debug every linux or windows desktop possible ! ;) 

For the record, just my feedback on this point : 
- stock ubuntu with unity -> ok
- remove unity and install only openbox and lxde -> fullscreen problem
- re-install unity -> ok under unity
- logout from unity and login to openbox or lxde -> fullscreen problem

martialgallorini, 2016-02-20:
well... it doesn't work... at all !
At least the previous solution made made the window size go fullscreen but keeping the window decoration. 

Here the window stays the size defined in main.cpp

kr15h, 2016-02-21:
Weird. But I'd say it is the way openFrameworks interacts with the desktop. What desktop environment are you using on Ubuntu?

martialgallorini, 2016-02-21:
I had this fullscreen problem under ubuntu 14.04 64 bits with LXDE for the first time

I am now under Debian Jessie 64 bits with Openbox. This is where i tested your new fullscreen settings. Let me know if you would like me to test under another distribution/environnement

For the record, this is the workaround i found : 

```
void ofApp::update(){
    if(!bFullscreen){
        float t = ofGetElapsedTimef();
        if (t > 5) {
            ofSetFullscreen(true);
            bFullscreen = true;
        }
    }
}
```

So waiting for some time (5 seconds) before throwing the app to fullscreen works...

It is like if using OF_FULLSCREEN mode on main() triggers fullscreen a bit too early, before something get initialised, but i can't understand what.

kr15h, 2016-02-21:
OK, nice. I will add this today. Yes, it might be that the window does a async or delayed interface init and thus when openFrameworks is demanding fullscreen, the window is not ready for it yet.

martialgallorini, 2016-02-21:
i tried 2 seconds but it seems a bit short, problem persists...

Also mouse hide doesn't work either at launch. See #47

martialgallorini, 2016-02-22:
bug reported to openFrameworks team.
was a bug from their part.

https://github.com/openframeworks/openFrameworks/issues/4848

now fixed. Thanks a lot

kr15h, 2016-02-22:
Cool! Waiting for the next oF release then.

closed 2016-02-22 by kr15h