kr15h/ofxPiMapper #44: Very sensitive mouse behaviour

magdesign, 2016-02-13:
Using ofxPiMapper without X.

When using generic USB mouse, its very sensitive to use, is it possible to adjust the mouse sensitivity somewhere ?

Also Logitech wireless mouse is not working, where is the location to load mouse drivers ?

mojibaked, 2016-03-01:
If it's an option for you, have you tried increasing the Raspberry screen resolution?
This helped in my case.
In /boot/config.txt, uncomment the lines hdmi_group and hdmi_mode, and change hdmi_mode=1 to hdmi_mode=16
This forces the hdmi output resolution to 1920x1080px.

magdesign, 2016-03-01:
Thanks for the trick, already running in full hd.
But since Kris implemented remote, I'm now building a remote to use mapper without mouse,
I guess this is the simplest and best solution.

closed 2016-03-01 by magdesign