kr15h/ofxPiMapper #46: Can I use an X Display as a source?

Taconut, 2016-02-20:
I have a Raspberry Pi and a small pocket-sized projector and I'd like to be able to project my raspberry pi to any wall while correcting for keystone distortion. Is there any way to mirror my desktop using this program?

kr15h, 2016-02-20:
It should work if you use 2 Raspberry Pi's - one booted up in desktop mode and another with [this](http://www.auvidea.eu/index.php/theme-styles/2014-12-30-22-32-06/b101) Raspberry Pi camera replacement board connected to it.

Like this:
[image]

Taconut, 2016-02-21:
Any single-device way to do it?

kr15h, 2016-02-21:
No.

Taconut, 2016-02-21:
Alright. I'll look into using an alternative solution and I'll keep the frame buffer idea in mind. Thanks.

closed 2016-02-24 by kr15h