kr15h/ofxPiMapper #50: Ogg support ?

martialgallorini, 2016-02-29:
Hello,

I would like to use the Ogg codec form my videos as it appears to be the smoother solution for me.

Is there a way to make Ogg codec usable ?

I think ofxPiMapper uses GStreamer which should support the Ogg Theora codec, but in ofxPiMapper example, when switching to media selection mode (mode 4), ogg encoded videos doesn't show up.

Thanks

kr15h, 2016-02-29:
Hi! Currently ofxOMXPlayer is used to play videos. OMX supports H264. The default ofVideoPlayer uses GStreamer, but recently I discovered that it has serious memory problems - loading and unloading and loading videos again eat up memory, there is a memory leak somewhere. So for now I would suggest you to use H264.

martialgallorini, 2016-02-29:
OK. But i am not using it on Raspberry Pi so OMX can't be used.
I will use another codec though, but would be great to see Ogg Theora ability.

thanks a lot.

kr15h, 2016-02-29:
Then theoretically it should fall back to ofVideoPlayer and if it does not support OGG, then...
I use `#ifdef TARGET_RASPBERRY_PI` to determine whether one is compiling on a Pi or elsewhere. If it is not a Pi, ofVideoPlayer is used. ofxOMXPlayer otherwise.

martialgallorini, 2016-02-29:
Yes i understand that.
But i was wondering if it is really an ofVideoPlayer problem or just a directory listing issue because .ogv doesn't even show up in the video list. I don't know how this list is built (by file type ? by file extension ? ...).

kr15h, 2016-02-29:
Yes, that's possible. Will check it.

commit 8bde5b84 (2016-03-01): Add `.ogv` file format, close #50
closed 2016-03-01 by kr15h