kr15h/ofxPiMapper #52: error building in debian 64 bits

martialgallorini, 2016-02-29:
hello,

here i am facing these errors with the latest commit : 

`In file included from /home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp:1:0:
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.h:29:48: error: expected class-name before ‘{’ token
 class TerminalListener : public Poco::Runnable {
                                                ^
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.h:42:9: error: ‘Thread’ in namespace ‘Poco’ does not name a type
   Poco::Thread thread;
         ^
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp: In member function ‘void ofx::piMapper::TerminalListener::setup(ofx::piMapper::KeyListener*, int)’:
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp:37:8: error: expected unqualified-id before ‘.’ token
  thread.start(*this);
        ^
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp: In member function ‘void ofx::piMapper::TerminalListener::run()’:
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp:41:15: error: expected primary-expression before ‘.’ token
  while (thread.isRunning()){
               ^
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp:62:10: error: expected unqualified-id before ‘.’ token
    thread.sleep(sleepTime);
          ^
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp: In member function ‘void ofx::piMapper::TerminalListener::close()’:
/home/major/oF092/addons/ofxPiMapper/src/Utils/TerminalListener.cpp:69:8: error: expected unqualified-id before ‘.’ token
  thread.tryJoin(50);
        ^
/home/major/oF092/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:335: recipe for target '/home/major/oF092/addons/obj/linux64/Release/ofxPiMapper/src/Utils/TerminalListener.o' failed
make[1]: *** [/home/major/oF092/addons/obj/linux64/Release/ofxPiMapper/src/Utils/TerminalListener.o] Error 1
make[1]: Leaving directory '/home/major/oF092/apps/myApps/ofMapper'
/home/major/oF092/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:126: recipe for target 'Release' failed
make: *** [Release] Error 2
`

building an old branch is ok. For example this one : 
https://github.com/kr15h/ofxPiMapper/tree/53b7d31c976ff0c26e1272d786595d14a09999e8

kr15h, 2016-02-29:
OK, will look into this.

martialgallorini, 2016-03-02:
ok great. Does it build on your side !?

kr15h, 2016-03-02:
Yes, it does on Pi 2 with Raspbian Jessie and oF 0.9.2.

kr15h, 2016-03-02:
Maybe try to run `make clean` and then `make` again

martialgallorini, 2016-03-02:
hum... strange... i definitely can't build it under Jessie 32bits or 64bits even on fresh installs...

But i am on intel platform, not arm... i don't think it is that different... but maybe it is.

closed 2016-04-18 by kr15h