kr15h/ofxPiMapper #53: Raspberry PI 3 + textured 1080p video?

mojibaked, 2016-03-01:
I've tried running ofxPiMapper on the new RPI 3 with Raspbian Jessie Lite, and it works perfectly!
1080p video in textured mode runs at 30-40fps, but still doesn't run as smooth as i expected.
There's still some horizontal tearing, with a 1500kbps 1080p mp4 file.
Tested with GPU memory 128 - 256 - 320 - 768 mb.

RPI 3 only came out yesterday, but i'm very keen to know if this is possible with the new hardware!

kr15h, 2016-03-01:
Wow, you're fast : ] I'm still only pulling myself together to get one of the 3's. But that's awesome!

magdesign, 2016-03-01:
Did you use a video with an audio stream ?
Does it run smooth in ofxOmxplayer without mapper ?
Check the readme here, https://github.com/jvcleave/ofxOMXPlayer

Unbelievable haw fast you got an Rpi3 :-)

mojibaked, 2016-03-01:
Yes, video without audio stream.
Haven't tried running it in "plain" ofxOMXPlayer. This morning i ran a ofxPiMapper binary (compiled on RPI2) on the RPI3. It worked out of the box which i didn't expect :)
So need to compile openFrameworks first, and install ofxOMXPlayer.
I'll try to do that tomorrow if i have the chance.
Will keep you updated!

mojibaked, 2016-03-04:
Had to wait a few days for a 2.5a power adapter and better SD card.
Just compiled openFrameworks 0.9.2 on a fresh Raspbian Jessie installation, and installed both ofxPiMapper and ofxOMXPlayer.
The examples work right out of the box!
Running the example-basic (no shaders) from ofxOMXPlayer with a 1080p mp4 video also causes horizontal tearing, with ±25-50 frames per second. 
In the example-basic, the tearing happens in both large and small frame simultaneously/equally.
GPU memory set to either 320 or 512mb

kr15h, 2016-03-04:
Cool. I managed to successfully compile openFrameworks on Pi 3 on Wednesday. Time was too short for trying to compile ofxPiMapper. Good to know that it works. Added duplicate surface feature just now.

closed 2016-04-18 by kr15h