kr15h/ofxPiMapper #56: Something big is wrong (problems with mouse, fullscreen, info.. etc.)

kokashking, 2016-05-09:
Hi Krisjanis,

first of all thank you for this great project!!! 

Last weekend I spent trying to get ofxPiMapper working as expected. But I didn't get it. It felt like I make something totally wrong like I would use incompatible version of OF or os. No mouse (doesn't matter wether it is the projection mapping state), fullscreen doesn't work, no info is showed (only grey lines), the right surface is empty and etc...  
But I use: RPI 2B, Raspbian Jessie Lite, OF 0.9.2.. and actually it should work. On MacOS ofxPiMapper works as expected.

May be you have seen something similar and can directly recognize what is wrong...

Thank you in advance. 

here are the output and the pic of the fullscreen mode:
```
pi@raspberrypi:~/openFrameworks/addons/ofxPiMapper/example $ make run
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 
[warning] ofAppEGLWindow: init(): X11 not availble on RPI yet, using a native window instead
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1920x1080
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 800x450
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface 
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): mouse_fd= 5 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.2:1.0-event-mouse
[notice ] ofAppEGLWindow: setupMouse(): mouse device name = Razer Razer Copperhead Laser Mouse
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard_fd= 6 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.5.2:1.0-event-kbd
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard device name = Apple Inc. Apple Keyboard
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[notice ] MediaServer: Attempting to add FBO source with name Cross FBO Source
[notice ] MediaServer: Source new, adding
[notice ] MediaServer: Attempting to add FBO source with name Custom FBO Source
[notice ] MediaServer: Source new, adding
[ error ] Application::loadXmlSettings(): surfaces.xml does not exist
[warning] Application::setup(): Failed to load user settings

[notice ] MediaServer: Attempting to load FBO source with name Custom FBO Source
[notice ] FboSource: Adding app listeners
[notice ] MediaServer: Current Custom FBO Source reference count: 1
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 50 MS
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 33.3333
[notice ] bool OMXClock::OMXStart(double, bool): at pts: 0
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[notice ] MediaServer: Initialized reference count of /home/pi/openFrameworks/addons/ofxPiMapper/example/bin/data/sources/videos/test.mov to 1
[notice ] MediaServer: Initialized reference count of /home/pi/openFrameworks/addons/ofxPiMapper/example/bin/data/sources/images/image4.jpg to 1
```

[image]

magdesign, 2016-05-09:
Hi kokashking 
About the mouse, I figured out that ofxPiMapper is not working with a Bluetooth/Wifi mouse. Did you try it with using an oldschool USB cable mouse? I guess same thing with the keyboard.

kokashking, 2016-05-09:
Hi Marc,

thank you for a quick reply but I'm using wired versions of mouse and keyboard. It is good to know for the future setups, that everything should be wired.

kokashking, 2016-05-10:
found the problem... while installing openframeworks I executed the tips from [openframeworks raspi getting-started](http://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/):

> We need to make sure the CPU has 192MB of RAM in order to compile openFrameworks. Once you have compiled openFrameworks you may want to repeat this step with 128

> Select 1 Expand Filesystem and hit Enter
>Select 8 Advanced Options and hit Enter
>Select A3 Memory Split and hit Enter
>Type 64 and Hit <ok>

I did it, though it was not even required. I own RPI 2B, therefore it would have enough memory. ))

the [ofxOMXPlayer](https://github.com/jvcleave/ofxOMXPlayer) in its turn requires 128mb for GPU, so I just needed to switch it back to 128.

closed 2016-05-10 by kokashking