kr15h/ofxPiMapper #57: Issue with deleting surface

awilkie1, 2016-05-15:
When trying to remove a surface through SSH or main keyboard the surface cannot be deleted. Have tried a variety of ways including the latest version of PIMapper. Possibly because of a PI update ??

kr15h, 2016-05-16:
Hi. I guess it's not implemented yet. The problem is that via SSH the backspace key does not work and thus I am trying to figure out what could be the replacement key.

awilkie1, 2016-05-16:
Thanks for getting back to me. it also isnt working with the keyboard plugged in.

kr15h, 2016-05-16:
Do the other shortcuts work? When you plug in the keyboard, you should restart the pi and the ofxPiMapper software for the keyboard to be properly recognised. Then go to mode 3, select a surface and hit BACKSPACE. Should work.

awilkie1, 2016-05-16:
just tested it. All other shortcuts work apart from the back space. checked keyboard as well thats fine! Was working untill i did upgrade & update, so i guess some modification to raspbian have caused an issue?

magdesign, 2016-05-16:
Please use 'd'  as replacement for backspace to delete surface... 
'a' for duplicate...
here a list of shortcuts I recommend (since its easier to simulate regular keys instead of special characters): https://github.com/magdesign/PocketVJ-CP-v3/files/188104/Keys_PocketVJ-VP-v3.pdf

kr15h, 2016-05-16:
I have to think of a more decent replacement as the SSH remote anyway. It crashes sometimes. Not good. But a shortcut convention would be nice. I'd also prefer to use ctrl keys. ctrl+d most of the time is used for duplication. ctrl+a for select all. etc.

awilkie1, 2016-05-16:
Ok great so its a key issue. i will change them in the project! Thanks

 p.s would be great to get TCPSyphon Client in there at some point as a input file

kr15h, 2016-05-16:
Yes! TCPSyphon would be that awesome!

kr15h, 2016-05-21:
By the way I see that via SSH one has to use "-" in order to delete. @awilkie1 did you try that?

kr15h, 2016-05-21:
@magdesign I disagree with using 'd' for delete, the key is conventionally used in other software to duplicate. And I want to reserve 'a' for "select all" or similar.

awilkie1, 2016-05-21:
Im not using ssh, just strait keyboard. However its become glitchy and sometimes works and sometimes not. I think adding some gui button would solve this within the application and would speed mapping up.

kr15h, 2016-05-21:
Ah. OK. Will check that. Does your keyboard have a DEL button? Another shortcut would be "x" which is a delete button also in Blender. That could be an alternative.

kr15h, 2016-06-06:
Are you still having problems?

closed 2016-06-07 by awilkie1