kr15h/ofxPiMapper #59: Refactor to avoid touching global GL states.

JoshuaBatty, 2016-05-25:
Hi mate just trying to use your addon at the moment and have run into a problem. In your BaseSurface class you call ofEnableNormalisedTexCoords() and then I see that in your example you need to manually call ofDisableNormalisedTexCoords in each draw loop. Although this works in this instance I think it's best to avoid setting any GL global state within ofxPiMapper. It creates a lot of headaches to manually take are of what the current state is when implementing the addon into larger projects. 

Do you have any thoughts on this? Cheers.

kr15h, 2016-05-25:
Hey! Thanks for pointing out. Will try to look into that. Although `ofEnableNormalisedTexCoords()` is needed in order the texture mapping to work on the pi.

kr15h, 2016-08-11:
Fixed. https://github.com/kr15h/ofxPiMapper/commit/c681e2e8b07033955c9f8210831aaad1b6ed3b17

closed 2016-08-11 by kr15h

JoshuaBatty, 2016-08-11:
Awesome thanks dude i'll give this another run this week. Cheers.