kr15h/ofxPiMapper #60: Unable to compile example-camera, 2 errors

yanghexiang, 2016-06-02:
Hello,

Am looking to source my own HD webcam to map. However, I cannot even get to compile the example-camera project. The original example project works fine.

here is what happens when i typed in make:

```
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 
Compiling OF library for Release
make[1]: Entering directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 
Done!
make[1]: Leaving directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'

Compiling example-camera for Release
make[1]: Entering directory '/home/pi/openFrameworks/addons/ofxPiMapper/example-camera'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 
Compiling /home/pi/openFrameworks/addons/ofxPiMapper/example-camera/src/CameraSource.cpp
g++ -c -O3 -DNDEBUG -Wall -std=c++14 -DGCC_HAS_REGEX -march=armv6 -mfpu=vfp -mfloat-abi=hard -fPIC -ftree-vectorize -Wno-psabi -pipe -DOF_USING_GTK -DOF_USING_GTK -DTARGET_RASPBERRY_PI -DSTANDALONE -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -I/home/pi/openFrameworks/addons/ofxOMXPlayer/src -I/home/pi/openFrameworks/addons/ofxOMXPlayer/libs/ffmpeg/include -fPIC -U_FORTIFY_SOURCE -Wall -ftree-vectorize -ftree-vectorize -Wno-deprecated-declarations -Wno-sign-compare -Wno-unknown-pragmas -Wno-deprecated-declarations -Wno-sign-compare -Wno-unknown-pragmas -Wno-unused-function -Wno-unused-but-set-variable  -I/opt/vc/include -I/opt/vc/include/IL -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -D_REENTRANT -pthread -I/usr/include/gstreamer-1.0 -I/usr/include/AL -I/usr/include/alsa -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng12 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/home/pi/openFrameworks/libs/fmodex/include -I/home/pi/openFrameworks/libs/glfw/include -I/home/pi/openFrameworks/libs/glfw/include/GLFW -I/home/pi/openFrameworks/libs/kiss/include -I/home/pi/openFrameworks/libs/poco/include -I/home/pi/openFrameworks/libs/tess2/include -I/home/pi/openFrameworks/libs/utf8cpp/include -I/home/pi/openFrameworks/libs/utf8cpp/include/utf8 -I/home/pi/openFrameworks/libs/openFrameworks -I/home/pi/openFrameworks/libs/openFrameworks/utils -I/home/pi/openFrameworks/libs/openFrameworks/video -I/home/pi/openFrameworks/libs/openFrameworks/3d -I/home/pi/openFrameworks/libs/openFrameworks/sound -I/home/pi/openFrameworks/libs/openFrameworks/communication -I/home/pi/openFrameworks/libs/openFrameworks/types -I/home/pi/openFrameworks/libs/openFrameworks/math -I/home/pi/openFrameworks/libs/openFrameworks/events -I/home/pi/openFrameworks/libs/openFrameworks/graphics -I/home/pi/openFrameworks/libs/openFrameworks/app -I/home/pi/openFrameworks/libs/openFrameworks/gl -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/example-camera.xcodeproj/xcuserdata -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/example-camera.xcodeproj/xcuserdata/kr15h.xcuserdatad -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/example-camera.xcodeproj/xcuserdata/kr15h.xcuserdatad/xcschemes -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/example-camera.xcodeproj/xcshareddata -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/example-camera.xcodeproj/xcshareddata/xcschemes -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/example-camera.xcodeproj/project.xcworkspace -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/example-camera.xcodeproj/project.xcworkspace/xcshareddata -I/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/src -I/home/pi/openFrameworks/addons/ofxPiMapper/src -I/home/pi/openFrameworks/addons/ofxPiMapper/src/Sources -I/home/pi/openFrameworks/addons/ofxPiMapper/src/Application -I/home/pi/openFrameworks/addons/ofxPiMapper/src/UserInterface -I/home/pi/openFrameworks/addons/ofxPiMapper/src/Info -I/home/pi/openFrameworks/addons/ofxPiMapper/src/Commands -I/home/pi/openFrameworks/addons/ofxPiMapper/src/Surfaces -I/home/pi/openFrameworks/addons/ofxPiMapper/src/MediaServer -I/home/pi/openFrameworks/addons/ofxPiMapper/src/Utils -I/home/pi/openFrameworks/addons/ofxOMXPlayer/src -I/home/pi/openFrameworks/addons/ofxOMXPlayer/libs/ffmpeg/include -I/home/pi/openFrameworks/addons/ofxIO/src -I/home/pi/openFrameworks/addons/ofxIO/libs -I/home/pi/openFrameworks/addons/ofxIO/libs/snappy -I/home/pi/openFrameworks/addons/ofxIO/libs/snappy/src -I/home/pi/openFrameworks/addons/ofxIO/libs/alphanum -I/home/pi/openFrameworks/addons/ofxIO/libs/alphanum/include -I/home/pi/openFrameworks/addons/ofxIO/libs/json -I/home/pi/openFrameworks/addons/ofxIO/libs/json/license -I/home/pi/openFrameworks/addons/ofxIO/libs/json/include -I/home/pi/openFrameworks/addons/ofxIO/libs/lz4 -I/home/pi/openFrameworks/addons/ofxIO/libs/lz4/src -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO/include -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO/include/ofx -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO/include/ofx/IO -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO/src -I/home/pi/openFrameworks/addons/ofxIO/libs/alphanum/include -I/home/pi/openFrameworks/addons/ofxIO/libs/json/include -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO/include -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO/include/ofx -I/home/pi/openFrameworks/addons/ofxIO/libs/ofxIO/include/ofx/IO -I/home/pi/openFrameworks/addons/ofxXmlSettings/src -I/home/pi/openFrameworks/addons/ofxXmlSettings/libs -I/home/pi/openFrameworks/addons/ofxGui/src -I/home/pi/openFrameworks/addons/ofxRPiCameraVideoGrabber/src -MMD -MP -MF obj/linuxarmv6l/Release/src/CameraSource.d -MT obj/linuxarmv6l/Release/src/CameraSource.o -o obj/linuxarmv6l/Release/src/CameraSource.o -c /home/pi/openFrameworks/addons/ofxPiMapper/example-camera/src/CameraSource.cpp
/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/src/CameraSource.cpp: In constructor 'CameraSource::CameraSource()':
/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/src/CameraSource.cpp:13:22: error: 'class OMXCameraSettings' has no member named 'isUsingTexture'
   _omxCameraSettings.isUsingTexture = true;
                      ^
/home/pi/openFrameworks/addons/ofxPiMapper/example-camera/src/CameraSource.cpp:16:23: error: 'omxCameraSettings' was not declared in this scope
   _videoGrabber.setup(omxCameraSettings);
                       ^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/CameraSource.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/CameraSource.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxPiMapper/example-camera'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2
```

kr15h, 2016-06-03:
Yes, the camera example needs extra attention indeed. Will look into this.

yanghexiang, 2016-06-03:
Thanks for the reply kr.

Is it feasible to combine videoGrabber with the original example?

I'm going to try it. If you have any suggestions that would be great.

kr15h, 2016-06-03:
Should be. On the Pi the camera example uses ofxRPiCameraVideoGrabber

kr15h, 2016-06-03:
I will check the example today/tomorrow hopefully and let you know here.

yanghexiang, 2016-06-03:
Hey that will be great. Appreciate that.

closed 2016-06-06 by kr15h

magdesign, 2018-04-26:
Trying to compile example_camera
Copied all files from _/openFrameworks/addons/ofxRPiCameraVideoGrabber/src/_
to: _/openFrameworks/addons/ofxPiMapper/example_camera/src/_

But compiling terminates with a bunch of errors like this:
_‘OMX_ERRORTYPE FlushOMXComponent(OMX_HANDLETYPE, int)’ defined but not used [-Wunused-function]
 OMX_ERRORTYPE FlushOMXComponent(OMX_HANDLETYPE handle, int port)_

Is the example_camera still maintained?

kr15h, 2018-04-26:
I think I forgot to add `ofxRPiCameraVideoGrabber` to `addons.make`.

kr15h, 2018-04-26:
Fixed with f0ceeb4870201bcaaded5cb1c0e96c2f6aa6a7f1

magdesign, 2018-04-26:
still something buggy:

/home/pi/openFrameworks/addons/ofxOMXPlayer/src/ofxOMXPlayerSettings.h:8:7: error: previous definition of ‘class DirectDisplayOptions’
 class DirectDisplayOptions
       ^
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:199: recipe for target 'obj/linuxarmv6l/Release/src/main.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/main.o] Error 1

magdesign, 2018-04-27:
Nevermind.
I deleted all the folders and cloned them fresh, now it compiles.
Thanks for your help.

kr15h, 2018-04-27:
OK great!

commit cdb4dc99 (2016-06-06): Fix example-camera not compiling on the Pi, close #60