kr15h/ofxPiMapper #63: Simplify README

kr15h, 2016-06-06:
The readme is unreadable. Too much information.

miharix, 2017-10-16:
Not to much, to little information.

I'm writing down all steps, so kids in my school club could replicate.
Unfortunately I'm making something wrong. 
When I run ./example I get bunch of
`[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb0/""`
 (for usb0 to usb3)
Then the graphic window opens, but it doesn't fill the full screen, pressing "f" changes nothing.

Can some one spot where I make the fatal mistake/s ?
_(Later If kr15h want, I can help and write(& record?) the tutorial for using ofxPiMapper on RsPi)_

---
```
//Rasspberry Pi 2B steps. 15.10.2017

http://downloads.raspberrypi.org/raspbian/images/raspbian-2015-09-28/
download 2015-09-24-raspbian-jessie.zip //yes old Raspian Jessie
df -h
umount /dev/sde1
sudo dd bs=4M if=2015-09-24-raspbian-jessie.img of=/dev/sde

//waaaait...
//insert SD in RsPi + keyboard + Mouse + LCD + internet

sudo raspi-config
-> eabl ssh
-> Localisation options
--> timezone
--> keyboard
-> expand filesystem
-> memory split 128
-> Boot options
--> Desktop / CLI
---> Console Autologin
->reboot yes

sudo apt-get update && sudo apt-get -y upgrade
//waaaait..

//ssh from PC to RsPi
cd ~
wget http://openframeworks.cc/versions/v0.9.8/of_v0.9.8_linuxarmv6l_release.tar.gz
mkdir openFrameworks
tar vxfz of_v0.9.8_linuxarmv6l_release.tar.gz -C openFrameworks --strip-components 1
cd /home/pi/openFrameworks/scripts/linux/debian

sudo ./install_dependencies.sh
//waaaait.. ~30min !!!

sudo apt-get autoremove

cd /home/pi/openFrameworks/addons

git clone https://github.com/jvcleave/ofxOMXPlayer.git

cd /home/pi/openFrameworks/addons/ofxOMXPlayer
git checkout 0.9.0-compatible

git clone https://github.com/kr15h/ofxPiMapper.git

cd /home/pi/openFrameworks/addons/ofxPiMapper/example
make 

//waaaaaiiittt multiple hours !
//???? warning at all compiling files "variable 'ok' set but not used"

cd /home/pi/openFrameworks/addons/ofxPiMapper/example/bin

./example
```

kr15h, 2017-10-16:
Hello. Valid complaint. Don't  mind the errors. If a window open's, you are fine. In order to launch in fullscreen mode, add the -f flag. Like this.

```
./example -f
```

Regarding usb errors. It is a temporary solution for reading and copying files from USB drives. For it to work you have to install usbmount.

```
sudo apt-get install usbmount
```

With this you can copy media on the root of your USB flash memory and once the ofxPiMapper example gets started, it will copy relevant files to the sources directory.

Video tutorial. Behold. I am close. 

I also just made a freash image, go to https://ofxpimapper.com and click on Download Latest.