kr15h/ofxPiMapper #67: Best resolution-codec for a lot of videos...

pietrondo, 2016-06-07:
Hi, what is the best resolution/codec for a good projector mapping with almost different videos?

thx

kr15h, 2016-06-07:
1080 or 720p, h264 codec. Check the compression recomendations here in the README: https://github.com/jvcleave/ofxOMXPlayer

pietrondo, 2016-06-07:
> COMPRESSION RECOMMENDATIONS:

    Use MPEGStreamclip (http://www.squared5.com/)
    H.264
    Quality 50%
    PCM Audio (more compatible with HDMI)

And about linux users?
Which program need to use to compress the video?
At the moment we tried the 1280x800 with h264 for 10 videos.. not good idea!

kr15h, 2016-06-08:
I believe ffmpeg is the best choice. One can find GUI for it.

kr15h, 2016-06-08:
on Debian one would install it with `sudo apt-get install ffmpeg`. I have not used any GUI for it and I don't know what would be the best choice, but to encode for RPi you could use this line:
```
ffmpeg -i yourvideo.mov -c:v libx264 -preset medium -c:a libfdk_aac -b:a 128k yourvideo.mp4
```
Have not tried this, but it's not far from the truth.

pietrondo, 2016-06-08:
ok thx

pietrondo, 2016-06-08:
We are developping a visual installation with your software..
this is our first image 
[image]

but at the moment the video are very slow (1280x800). In the next days we try to change resolution (800x600) and add new "cubes".

kr15h, 2016-06-08:
Looks great! Thanks for sharing!

pietrondo, 2016-06-08:
Thx you. I'm working for a Italian Start-up and we do projector mapping.
With your software we could do more things.
How can we help you?
1)Translate documentation to italian?
2)Debug?
3)Test new improvmement?

thx very much

Pietro Capriata

http://www.pietrocapriata.me

CONFIDENTIALITY DISCLAIMER
Le informazioni contenute in questo messaggio di posta elettronica e negli
allegati sono riservate e confidenziali e ne sono vietate la lettura,
l'uso, la copia, la comunicazione e la diffusione in qualunque modo
eseguite. Qualora lei non fosse la persona destinataria del messaggio, la
invitiamo ad eliminarlo, dando gentilmente comunicazione al mittente
tramite e-mail di ritorno.

The information contained in this email message and any files transmitted
with it are confidential and privileged and any reading, processing,
distribution or copy of this material is strictly prohibited, in any form.
If you are not the intended recipient of this message, please immediately
delete it, giving the relevant communication to the sender by reply e-mail.

regaladys, 2016-09-27:
I have been playing 1080p HD videos with ofxPiMapper. The quality (some frames are skipped) is reduced compared to simply playing it with omxplayer. I assume this is expected as there is image/video processing involved in projection mapping unlike simple video output.

I'm curious what image/video processing is involved? Which parts of the code should I check out if I want to learn about it?

kr15h, 2016-09-27:
I'm using the ofxOMXPlayer for playing back videos on the Pi.

kr15h, 2016-09-27:
Hey @pietrondo ! Sorry for late reply. At the point I am restructuring the architecture step by step and there is nothing really you could help code-wise now. What I would appreciate though would be a couple of nice hi-res pictures, a paragraph of text what the project is about and a video that I could add to the showcase directory of ofxPiMapper. Don't forget to credit all the involved people in the description. I want to make a project showcase gallery some time in the near future.

regaladys, 2016-09-28:
Cool! If our Singapore exhibit goes well and uses ofxPiMapper, I'll send you some photos too @kr15h !

pietrondo, 2016-09-28:
I'll send them by email?

Pietro Capriata

http://www.pietrocapriata.me

CONFIDENTIALITY DISCLAIMER
Le informazioni contenute in questo messaggio di posta elettronica e negli
allegati sono riservate e confidenziali e ne sono vietate la lettura,
l'uso, la copia, la comunicazione e la diffusione in qualunque modo
eseguite. Qualora lei non fosse la persona destinataria del messaggio, la
invitiamo ad eliminarlo, dando gentilmente comunicazione al mittente
tramite e-mail di ritorno.

The information contained in this email message and any files transmitted
with it are confidential and privileged and any reading, processing,
distribution or copy of this material is strictly prohibited, in any form.
If you are not the intended recipient of this message, please immediately
delete it, giving the relevant communication to the sender by reply e-mail.

kr15h, 2016-09-28:
@pietrondo yes, that would be the best krisjanis at rijnieks com