kr15h/ofxPiMapper #68: Make interaction (request)

pietrondo, 2016-06-09:
Hello, is possible to interact with the videos?
For example, with a keyboard start and stop a specific video? 

thx

kr15h, 2016-06-09:
Will try to implement it in a simple way. Thanks for the request.

kr15h, 2016-06-10:
So just added the possibility to play/pause a video source once a surface is selected by using SPACE bar. Enjoy!

closed 2016-06-10 by kr15h

pietrondo, 2016-06-10:
ok thx.
Is possibile to "map" one specific surface with a button so you can play or stop with this specific button?

thx

kr15h, 2016-06-11:
Nope, now it is that you have to select the specific surface to play/stop

kr15h, 2016-06-11:
Well it is possible, but you have to add it to the source code. I will look into options how to make it configurable maybe.