kr15h/ofxPiMapper #69: New request. Remote sources?

pietrondo, 2016-06-18:
Possible new feature:
remote sources..
One pc transmit video and the raspberry output it on the video (in a square, triangle etc).
Possible?
thx

kr15h, 2016-06-19:
Yes. TCPSyphon source is one of the next things I am going to implement.

magdesign, 2016-12-01:
This could be done with TCPSyphon or with a more common protocol like RTP/UDP.
Syphon is nice when you have VJ software, RTP is cool because VLC can output it.

steve20pro, 2016-12-22:
adding to this another request, 
an ability to use an hdmi input as a source 

using a board such as this: http://auvidea.eu/index.php/theme-styles/2014-12-30-22-32-06/b102

pietrondo, 2017-04-12:
yes. Problem with TCPSyphon. It will work if you have a windows or mac os desktop.
If you have linux you can't output nothing (spout and syphon don't exist on linux)

thx for you software

hgshoggins, 2017-12-23:
Supporting remote sources would be absolutely wonderful.

If I'm correct, you're already using the FFMpeg API to access the video data, right ? If so, it looks like the possibility to access remote sources is part of it.

For example, I use OMXPlayer with remote sources simply by giving it an SDP file, and I successfully display an H264 RTP (UDP) stream with no effort. Looking into OMXPlayer's code, I couldn't find any evidence of SDP parsing, but found some in FFMPeg's source. As OMXPlayer is using FFMpeg to access the video, I assumed by deduction that it would be feasible in ofxPiMapper.

By the way, ofxPiMapper is just brilliant, thanks !