kr15h/ofxPiMapper #70: new ideas?

pietrondo, 2016-07-10:
1)Make a shell script witch automatize the install (and update) of ofxpimapper on raspbian
2)possible to select different maps at start-up
3)python scripting to interact with the program?

thx

magdesign, 2016-07-14:
- Rectangular meshes on grid warping (rows and culoms)
- Spherical surfaces
- Masking (like a simple surface which has black content to mask off some areas)
- Other FBO sources (nightrider style lines)
- Controls to adjust brightness, contrast, RGB

and so on :-)

kr15h, 2016-08-12:
Rect grid warp meshes, spherical surfaces and masking sounds fun.

magdesign, 2016-08-12:
maybe some bezier quads...: https://github.com/TsubokuLab/ofxBezierWarpManager

JoshuaBatty, 2016-10-10:
Ability to implement GUI of choice to control all the main interactive features of ofxPiMapper. It looks like maybe you are thinking about this already from what I can gather...? What are your thoughts on this? This feature is pretty much essential for integrating that addon into already established oF projects. Cheers.

kr15h, 2016-10-11:
Yeah, it is moving this way step by step.

JoshuaBatty, 2016-10-18:
@kr15h Great! would be awesome to have a basic example using ofGui, let me know if I can help.

kr15h, 2016-10-18:
Hey! First thing I want to do is to pass the mouse events down the pipe instead of using ofEvents in the addon. Then there would be an easy way to disable existing shortcuts. Next step would be to open up the internal API on the ofxPiMapper class level.

cgiles, 2016-11-14:
@kr15h taking the mouse event from the Linux event object is quite easy. I did something similar for the touchscreen official, and a joystick class for use a joystick on rpi with Openframeworks.  

I seen there is function related to preset in the code.  Did you implemented a multi scene system ?  That would be an interesting function.  

And an useful function would be to duplicate the surface shape, to the surface texture shape.  So it would be scaled.

magdesign, 2016-12-01:
Blending modes:
- add
- over
- multiply
- subtract

Bezier style handles on each corner

magdesign, 2016-12-13:
While going thru the xml setting files, I had an idea how to fix the gridwarp function, since its pretty useless in moment.
First,add a quad surface.
Now you get 2 coordinates for each corner point:

[image]

The new function would be called something like: Add vertical grid mesh
This would calculates the length between the points A+B and C +D, half it,
generate a new quad. Now we would have two quads, exactly at the same place as the one we had before, but on point E+F the quads are fixed/snapped together.
In the mesh input we would have the same behaviour, so it would be possible the move the vertical line of the input.

[image]

I have no idea how hard this would be to realise, but it would be a killer feature.
BTW. Also a Add horizontal grid mesh of course...

kr15h, 2016-12-13:
yes. i have ideas for improving this. behold : ]

magdesign, 2016-12-16:
New ideas.
Import function for .svg files.
I started to make a simple parser which converts .svg files exported from a closed source mapping tool to the PiMapper. The basic concept works (tested with .svg exported from Madmapper so far).
There is still an issue with the zero point, the created files have the zero point in the top left corner of the canvas instead of the center of the screen. So there must be a function to add half of the resolution to the values. Check it out here: https://github.com/magdesign/mapping-converter

kr15h, 2016-12-16:
Cool. Thanks. Will check it out as soon as possible. Kind of busy with projects right now, not so much time for development, but January looks promising.

magdesign, 2017-01-11:
Anyone out there who can code more FBO sources ?

kr15h, 2017-01-11:
You have some specific ideas or needs?

magdesign, 2017-01-11:
yea, i was thinking about:
- squares zooming in or out
- filling screen up from one side (to create animated lines)

still searching for motivated people to create some kind of opensource mapping community and push pimapper to a new level :-)

kr15h, 2017-01-11:
Yeah, more contributors would be cool..

magdesign, 2017-01-11:
Maybe we should organise a hackaton.....

kr15h, 2017-01-12:
Not a bad idea at all. Where and when?

magdesign, 2017-12-19:
We added a simple circular shape: https://github.com/magdesign/ofxPiMapper/tree/CircleSurface-no-RPi-shaders and the new slideshow, which you already adopted yesterday.
Since the implementation of slideshow FBO source, we figured out that ofxPiMapper runs the movies and any FBO sources regardless whether they are being actually used on a surface or not, this eats a lot of resources. Is there a way to manually turn them on/off or any other solution to save resources?

pietrondo, 2017-12-19:
good work @magdesign  I'll use your improvements in our next installation..

kr15h, 2017-12-20:
Cool. @magdesign there is a way of overriding the shortcuts by not changing them into ofxPiMapper codebase. I would be very happy if we could make another example, which would be specifically targeted for PocketVJ.

kr15h, 2017-12-20:
Therefore would be really cool if we could use pull-requests etc, etc

cgiles, 2017-12-23:
a fbo which can play streamed videos would be cool too :)

kr15h, 2017-12-23:
Yeah, I want to make it for a longer while now.

magdesign, 2017-12-23:
Is there a simple way to compile PiMapper without FBO sources to gain more performance?
(what do I have to comment out?)

about streaming: as soon as NDI for ARM is availabe, this might be interesting.

kr15h, 2017-12-23:
Oh, NDI would be badass.

kr15h, 2017-12-23:
In short, I will make an "example_pocketvj" without FBO sources and with PV shortcuts so you could add features using it as the sandbox.