kr15h/ofxPiMapper #71: Audio Playback

magdesign, 2016-08-03:
Is there any chance to enable audio playback to the minjack?

This would be amazing!
Or is it already available and I miss something?

kr15h, 2016-08-03:
I think this is what you have to do: https://gist.github.com/jvcleave/4972661

kr15h, 2016-08-03:
And `VideoSource::enableAudio = false;`

magdesign, 2016-08-04:
Thanks for the hint.
I changed in: example/src/ofApp.cpp
line 8: ofx::piMapper::VideoSource::enableAudio = true;

and in: src/Sources/VideoSource.cpp 
line 6:bool VideoSource::enableAudio = true;

Audio now plays back thru HDMi but sounds very glitchy... May be the change insrc/Sources/VideoSource.cpp is not necessary. Will figure out how to change to min jack, the modprobe solution does not have any effect. Will post results later.
Maybe there are some other tricks as well.....

kr15h, 2016-08-04:
It might be also audio encoding problem. Have you tried to play back the video from commandline using `omxplayer`?

kr15h, 2016-08-04:
The compiling is a bit of a mystery to me as it seems that it did compile faster on the older Raspberry Pi's . Maybe there is something that has been changed in oF. Might be interesting to dig into that. But crosscompiling setup is soo needed for me now. Have to work on that. Make a kind of a network-boot development platform.

magdesign, 2016-08-04:
Yea, audio is smooth with omxplayer, jack and hdmi.
We compile omxplayer with cross-compiling... sd cardreader on centoOS computer.. its very fast, will check if this would also work with ofx. 
Maybe there is an option to allocate all 4 cores before compiling on the pi.... its just such a pain to wit soo long after changing an option :-)

magdesign, 2016-08-07:
After messing with raspi-config, conf.txt, .asoundrc, amixer cset and alsa stuff, I figured out that there is a line in: src/Sources/VideoSource.cpp on line 28
which must be changed to: settings.useHDMIForAudio = false;

I'm not on the newest PiMapper version, so it might be that you don't have this line, then change it in: src/Sources/OMXPlayerCache.cpp 
on line 21

compile and after that, audio is working on jack.

The Audio quality is not the best, but for my purpose it works so far....

kr15h, 2016-08-07:
ok, maybe adding a config variable for this one would be a good idea..

magdesign, 2016-08-07:
jap, that would be amazing!!

kr15h, 2016-08-12:
So I added a simple flag that should be set before running `piMapper.setup()`. Use `VideoSource::useHDMIForAudio = true;` to enable HDMI audio output.

magdesign, 2016-12-20:
Is there a chance to output audio to an ALSA device ?

For example: VideoSource::usealsa:hw:1,0 = true

kr15h, 2016-12-20:
Good question. I think that due the changes in the ofxOMXPlayer API this is not working anymore.

magdesign, 2017-01-12:
Opened an issue on ofxOMXPlayer concerning ALSA support.
Hopefully jvcleave will update his fork of omxplayer.
https://github.com/jvcleave/ofxOMXPlayer/issues/101

magdesign, 2017-03-18:
Just figured out that you made a lot of changes in the code.
Seems to me that you dont use ofxOmxplayer anymore, its not listed in the dependencies anymore.
Does this mean its now possible to have ALSA support?
What videoplayer do you use now instead?

kr15h, 2017-03-18:
It is still there in the stable branch. I am developing the master branch to be compatible with the openFrameworks master branch now. The removal of ofxOMXPlayer has to be tested still. Will do this the following week. I removed the ofxIO and am removing the ofxOMXPlayer dependencies because most of the configuration problems from users are related to these addons.

kr15h, 2017-03-18:
And in the stable branch there are steps how to make the sound work. I mean. I tested it last week with alsamixer and omxplayer - it was working without problems. The trick might be that you should set the volume to 90 - 100% in alsamixer

magdesign, 2017-03-18:
Cool, thanks!
will test this,
great effort.

natcl, 2017-09-07:
Is it currently possible to output to a USB sound device ? I have a USB soundcard which is set to default in Raspbian but sound still plays from the built-in jack