kr15h/ofxPiMapper #72: Input Selection

magdesign, 2016-08-03:
If I want to select to input by mouse (the old fashioned way) is there a possibility to make the fonts bigger and especially the checkboxes inverted, e.g. white checkbox with black or even red cross?

I'm sure its possible to change this somewhere before compiling.

Thanks for a hint!

kr15h, 2016-08-03:
I think that the ofxGui does not offer that. I have great plans reworking the UI anyway. Great plans, no time haha. Have to solve some things first.

pietrondo, 2016-08-25:
or by a web interface?

kr15h, 2016-08-25:
yeah, that involves running a web server and building a not too snappy web interface. I have tried that and I would like to go for something that is as snappy as possible.