kr15h/ofxPiMapper #75: Add presets

kr15h, 2016-10-01:
Sometimes there is a need to create and switch between multiple mapping presets. In projection mapping mode there would be a possibility to do it.

doctea, 2017-05-29:
Seems like this functionality is working, but no UI keyboard shortcuts to use it..?  Are there any known bugs with it?  I've hacked in support for eraseActivePreset(), cloneActivePreset() and selectNextPreset() and it all seems to be loading, saving and switching OK as far as I can tell but there are still some TODOs regarding this left in the source..

kr15h, 2017-05-29:
Yes. There are some other things to solve before. I am reworking the source system currently. Want to make the source list update automatically without reboot and remove the mode 4 which uses ofxGui. No ofxGui. ofxGui not too good for the purpose.

doctea, 2017-05-29:
Cool stuff.  Thanks for all your hard work on this project!  I've got an event coming up at the end of July and I'm currently hoping to use ofxPiMapper for this if I can get things working how I want.  I was looking for a way to quickly switch between different surface/source configurations, in order to trigger changes 'live', and using the Presets functionality for this seems to be the quickest way for now.

kr15h, 2017-05-29:
Yes. That's the way for now.