kr15h/ofxPiMapper #77: Command line build not working on mac after checkout

tilmans, 2016-10-18:
The makefile will try to compile ofxOMXPlayer which won't work on mac (macos Sierra + of094). The build in XCode will work just fine. Maybe the README could mention that since I imagine a lot of people will first try to compile on their dev machine. 

Keep up the good work! 👍

kr15h, 2016-10-18:
Yes. I should add that. Maybe there is a workaround. Will check.

kr15h, 2016-12-01:
I think it is because the addon_config.mk file. It had ofxOMXPlayer as common addon dependencies, but ofxOMXPlayer is not common, it is only on Raspberry Pi. I removed it from the addon_config.mk. Please try again and let me know if it works for you now.

closed 2016-12-01 by kr15h

tilmans, 2016-12-01:
works now, thank you!