kr15h/ofxPiMapper #79: Error to Compile new videos

fjrobles, 2016-10-31:
[image]

I can´t compile new videos with new images everything is ok...

kr15h, 2016-10-31:
I see you are using a mov file. The Raspberry Pi supports H264 encoding. You should encode your video to MP4. Also with an mp4 extension.

fjrobles, 2016-11-01:
Sorry, but I tried all videos from ofxPiMapper/example/bin/data/sources/videos/.

[image]

[image]

kr15h, 2016-11-01:
I see there is a missing </surface> closing tag just after <properties></properties>

kr15h, 2016-11-01:
In his case the video is not the problem, but the validity of the xml file. The application fails to parse it as it is malformed.

fjrobles, 2016-11-01:
I have validate the XML file but still doesnt work... Actually the screen appear totally white

[image]

kr15h, 2016-11-01:
It might be the Raspberry Pi memory split at this point. Try to increase the amount of the memory consumed by the GPU by opening the config tool with `sudo raspi-config`, go to Advanced Settings and Memory Split. Enter 256 there, save and exit. Reboot.

fjrobles, 2016-11-01:
Yes!!!!!! the problem was Memory Split!!!!! Thanks a lot!!!!!!!

kr15h, 2016-11-01:
Cool! : ] Good luck with the rest then. Just ask if there are more problems/bugs. One shalt close this issue now....

closed 2016-11-01 by kr15h