kr15h/ofxPiMapper #80: field ‘_application’ has incomplete type ‘ofx::piMapper::Application’

nanu-c, 2016-11-18:
```
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/PresentationMode.h:8:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:13,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/ProjectionMappingMode.h: At global scope:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/ProjectionMappingMode.h:39:38: error: invalid use of incomplete type ‘class ofx::piMapper::ApplicationBaseMode’
 class ProjectionMappingMode : public ApplicationBaseMode {
                                      ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/PresentationMode.h:7:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:13,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Commands/SetApplicationModeCmd.h:10:7: note: forward declaration of ‘class ofx::piMapper::ApplicationBaseMode’
 class ApplicationBaseMode;
       ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:13:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/PresentationMode.h:14:33: error: invalid use of incomplete type ‘class ofx::piMapper::ApplicationBaseMode’
 class PresentationMode : public ApplicationBaseMode {
                                 ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/PresentationMode.h:7:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:13,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Commands/SetApplicationModeCmd.h:10:7: note: forward declaration of ‘class ofx::piMapper::ApplicationBaseMode’
 class ApplicationBaseMode;
       ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:14:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/TextureMappingMode.h:24:35: error: invalid use of incomplete type ‘class ofx::piMapper::ApplicationBaseMode’
 class TextureMappingMode : public ApplicationBaseMode {
                                   ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/PresentationMode.h:7:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:13,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Commands/SetApplicationModeCmd.h:10:7: note: forward declaration of ‘class ofx::piMapper::ApplicationBaseMode’
 class ApplicationBaseMode;
       ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:16:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/SourceSelectionMode.h:12:36: error: invalid use of incomplete type ‘class ofx::piMapper::ApplicationBaseMode’
 class SourceSelectionMode : public ApplicationBaseMode {
                                    ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/PresentationMode.h:7:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:13,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Commands/SetApplicationModeCmd.h:10:7: note: forward declaration of ‘class ofx::piMapper::ApplicationBaseMode’
 class ApplicationBaseMode;
       ^~~~~~~~~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:68:30: error: field ‘_application’ has incomplete type ‘ofx::piMapper::Application’
   ofx::piMapper::Application _application;
                              ^~~~~~~~~~~~
In file included from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Application/Modes/PresentationMode.h:7:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/ofxPiMapper.h:13,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/nanu/visuals/openFrameworks/addons/ofxPiMapper/example/src/ofApp.cpp:1:
/home/nanu/visuals/openFrameworks/addons/ofxPiMapper/src/Commands/SetApplicationModeCmd.h:9:7: note: forward declaration of ‘class ofx::piMapper::Application’
 class Application;

```

nanu-c, 2016-11-19:
solved with new git clone of of ;)

closed 2016-11-19 by nanu-c

kr15h, 2016-11-19:
Good, as I did a fresh compile yesterday and no sign of similar things.