kr15h/ofxPiMapper #86: ofxIO issue

awilkie1, 2017-01-22:
Just a quick one. They have upgraded ofxIO meaning the current version doesn't run with pi mapper. Found if I downgraded ofxIO 9.0 it worked fine. A possible update of pi mapper to accommodate the new ofxIO would be great! Thanks

kr15h, 2017-01-22:
I think you just go to `addons/ofxIO` and `git checkout stable`. Recompile. Should work.

kr15h, 2017-01-22:
On the other hand, I want to get rid of this dependency. More is less dependencies. This issue is messing up the minds of too many users lately.

awilkie1, 2017-01-22:
i have it working now on a lower version of ofxIO but yeah would be great to fix or remove it as a dependency. Thanks for the support!

awilkie1, 2017-01-23:
Sorry to re-open the issue however thought I would update you on this one. tried running with all current stable versions of addons as you said a got this error any ideas ??
[image]

kr15h, 2017-01-23:
weird. will look into this...

janisso, 2017-03-17:
Getting similar error when trying to make example code.
I am using RPi 3 with OF 0.9.8 for armv6l with Jesse Pixel

/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/main.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/main.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxPiMapper/example'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed

kr15h, 2017-03-17:
You should use the `stable` branch of the ofxPiMapper repository since I introduced changes in the repo structure. Now the master branch is developed to match the openFrameworks `master` branch. ofxIO is removed in the current stable branch and it should not cause problems when compiling.

janisso, 2017-03-17:
Awesome! Paldies!

kr15h, 2017-05-26:
There is no ofxIO dependency anymore.

closed 2017-05-26 by kr15h