kr15h/ofxPiMapper #89: Toggle layers

mandelbrotmedia, 2017-01-28:
Is it possible to toggle the layers on and off with a keyboard shortcut?

mandelbrotmedia, 2017-03-04:
Hi! 
Do you know what i mean?
Lets say i have 4 layers.
Layer 1 and 2 are in the foreground, layer 3 and 4 in the background. All layer have 100% opacity now. 
If i press keyboard "5" the opacity of layer 1 should switch from 100% to 0% and the layer disappears. If i press "5" again the opacity should switch from 0% to 100%. Same for keyboard "6" and layer 2. Do you think there is a solution? Or can you give me a hint?

kr15h, 2017-03-04:
I could add a feature like ofxPiMapper::toogleSurface(int i); which would hide / unhide a surface. Or an opacity method. ofxPiMapper::setOpacityForSurface(int surfaceIndex, float opacity); From there you could add your custom shortcuts. The question is whether you need it while working on the mapping or you need to save and load it again?

mandelbrotmedia, 2017-03-07:
this sounds good! my use-case is as following: i would use 8 layers. 2 in the background and 6 small surfaces in the foreground. the 2 background layers should be visible and the 6 foreground layers should be hided.  i do the mapping and the save it and move to the presentation mode. in this mode i would use the hide/unhide feature by pressing different keys for toggling the visibility.