kr15h/ofxPiMapper #91: import resolume arena save file

pietrondo, 2017-02-09:
Hi,
Another feature in the next future could be the possibility of import resolume arena (a vj software for projector mapping) into the ofxpimapper? 
thx

magdesign, 2017-02-10:
It would be much better to make an import function which will work with every mapping software instead of only one..
I already started to code a tool based on .svg export, which most mapping softwares support. Maybe you can help finishing it! check out here: https://github.com/magdesign/mapping-converter

kr15h, 2017-02-10:
Yes, basic SVG and mesh importing should be it.

magdesign, 2018-01-12:
How you mean pull request?
Its a standalone application on which you define input .svg and output .xml files....

kr15h, 2018-01-12:
Ah, it is a tool. Ok. Never mind : ]