kr15h/ofxPiMapper #93: processing and ofxPiMapper

pietrondo, 2017-03-03:
I'm learning processing and...
is possible to use processing sketch into ofxPiMapper (without encoded them into video?)

thanks!

kr15h, 2017-03-03:
Hi @pietrondo. No, sorry to dissapoint you, bu it is not. You should get into openFrameworks and port your Processing patch to C++

pietrondo, 2017-03-03:
@kr15h thx! So i need to "learning open frameworks?
For your experience is easy to learn (i don't know c++). Do you know some good books or tutorial?
thx

kr15h, 2017-03-04:
@pietrondo I think that http://openframeworks.cc/learning/ is a good starting point. Then there is a book: https://www.amazon.com/Mastering-openFrameworks-Creative-Coding-Demystified/dp/1849518041/