kr15h/ofxPiMapper #95: Blurry Font

architek1, 2017-03-08:
I was using PiMapper on (5) Raspberry pi 3's to try map a ultra-short throw Optoma projection and it had stationary text on the top of the vertical video. The text ended up being aliased and quite hard to read. Is this a PiMapper issue or HDMI out the Pi? Setup: Rpi 3, Jessie Lite, OFX .9.8. and PiMapper.

kr15h, 2017-06-06:
Might be you have to set higher DPI for the font and load it with higher font size. Could you share the code where you set up and draw the font?

architek1, 2017-06-06:
I didn't have any code displays the font. It was embedded in the photo that was to be displayed. Client made a last minute change on site at the time.

kr15h, 2017-06-06:
What size was the image?

magdesign, 2017-06-06:
Sounds more like an issue of the image compression and/or RPi resolution settings.
Also some ultra short throw projectors cause issues if they are not in a perfect angle to the wall.
Upload the image on a hoster so we can check it.

kr15h, 2017-06-06:
My guess would be that that the image was too big (like 2000x2000 or something) and was scaled down to 500x500 during the mapping process. The method is used to quad warp the texture does not take care of aliasing. Thus if your screen is 1920x1080, the image should be no more than 1000x1000px

closed 2018-05-15 by kr15h