kr15h/ofxPiMapper #97: problem compiling master branch

pietrondo, 2017-03-24:
Hi.
I'm trying compiling the master branch with git openframeworks on my raspbian.
I have those errors..

`/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::~ofTessellator()':
ofTessellator.cpp:(.text+0x28): undefined reference to `tessDeleteTess'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::init()':
ofTessellator.cpp:(.text+0x88): undefined reference to `tessNewTess'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::operator=(ofTessellator const&)':
ofTessellator.cpp:(.text+0x100): undefined reference to `tessDeleteTess'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::performTessellation(ofPolyWindingMode, ofMesh_<glm::tvec3<float, (glm::precision)0>, glm::tvec3<float, (glm::precision)0>, ofColor_<float>, glm::tvec2<float, (glm::precision)0> >&, bool)':
ofTessellator.cpp:(.text+0x13c): undefined reference to `tessTesselate'
ofTessellator.cpp:(.text+0x204): undefined reference to `tessGetVertexCount'
ofTessellator.cpp:(.text+0x214): undefined reference to `tessGetElementCount'
ofTessellator.cpp:(.text+0x29c): undefined reference to `tessGetVertices'
ofTessellator.cpp:(.text+0x2cc): undefined reference to `tessGetElements'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::tessellateToMesh(ofPolyline_<glm::tvec3<float, (glm::precision)0> > const&, ofPolyWindingMode, ofMesh_<glm::tvec3<float, (glm::precision)0>, glm::tvec3<float, (glm::precision)0>, ofColor_<float>, glm::tvec2<float, (glm::precision)0> >&, bool)':
ofTessellator.cpp:(.text+0x414): undefined reference to `tessAddContour'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::tessellateToMesh(std::vector<ofPolyline_<glm::tvec3<float, (glm::precision)0> >, std::allocator<ofPolyline_<glm::tvec3<float, (glm::precision)0> > > > const&, ofPolyWindingMode, ofMesh_<glm::tvec3<float, (glm::precision)0>, glm::tvec3<float, (glm::precision)0>, ofColor_<float>, glm::tvec2<float, (glm::precision)0> >&, bool)':
ofTessellator.cpp:(.text+0x4d0): undefined reference to `tessAddContour'
ofTessellator.cpp:(.text+0x568): undefined reference to `tessAddContour'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::performTessellation(ofPolyWindingMode, std::vector<ofPolyline_<glm::tvec3<float, (glm::precision)0> >, std::allocator<ofPolyline_<glm::tvec3<float, (glm::precision)0> > > >&, bool)':
ofTessellator.cpp:(.text+0x5c0): undefined reference to `tessTesselate'
ofTessellator.cpp:(.text+0x688): undefined reference to `tessGetVertices'
ofTessellator.cpp:(.text+0x694): undefined reference to `tessGetElements'
ofTessellator.cpp:(.text+0x6a0): undefined reference to `tessGetElementCount'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::tessellateToPolylines(ofPolyline_<glm::tvec3<float, (glm::precision)0> > const&, ofPolyWindingMode, std::vector<ofPolyline_<glm::tvec3<float, (glm::precision)0> >, std::allocator<ofPolyline_<glm::tvec3<float, (glm::precision)0> > > >&, bool)':
ofTessellator.cpp:(.text+0xa5c): undefined reference to `tessAddContour'
/home/pietro/openframeworks/libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofTessellator.o): In function `ofTessellator::tessellateToPolylines(std::vector<ofPolyline_<glm::tvec3<float, (glm::precision)0> >, std::allocator<ofPolyline_<glm::tvec3<float, (glm::precision)0> > > > const&, ofPolyWindingMode, std::vector<ofPolyline_<glm::tvec3<float, (glm::precision)0> >, std::allocator<ofPolyline_<glm::tvec3<float, (glm::precision)0> > > >&, bool)':
ofTessellator.cpp:(.text+0xb18): undefined reference to `tessAddContour'
ofTessellator.cpp:(.text+0xbb0): undefined reference to `tessAddContour'
collect2: error: ld returned 1 exit status
/home/pietro/openframeworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:403: recipe for target 'bin/example' failed
make[1]: *** [bin/example] Error 1
make[1]: Leaving directory '/home/pietro/openframeworks/addons/ofxPiMapper/example'
/home/pietro/openframeworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2
`

pietrondo, 2017-03-24:
uname -a
Linux raspberrypi 4.4.27-v7+ #2 SMP Wed Oct 26 19:09:12 UTC 2016 armv7l GNU/Linux

kr15h, 2017-03-24:
Try to pull and recompile. Try also pull the latest changes from the openFrameworks master branch. But for production I would suggest to use the stable branch as I have not figured out a way to make hardware accelerated video work without using ofxOMXPlayer.

pietrondo, 2017-03-24:
ok thx (I need to make an Installation here and I need "something stable")
thx very much

kr15h, 2017-03-24:
And use the stable openFrameworks as well. 0.9.8.

kr15h, 2017-05-26:
Made master branch as stable branch again. Compatible with of 0.9.8. Developing against stable branch of openFrameworks.

closed 2017-05-26 by kr15h