kr15h/ofxPiMapper #103: selects more point in Surface editing mode

pietrondo, 2017-04-24:
Hi!
Sorry for my continue requests... but we are making an installation and we have a question. In the next release would be possible to select more point (so you can move in contemporaney two o more of them?)
thx

kr15h, 2017-04-25:
Hey! I know what you mean. Would love to have this feature myself. It's going slow, but features are coming ; ]

kr15h, 2018-02-07:
@JoshuaBatty Wow multitouch for ofxPiMapper. Yes, it has to be rewritten. If you have a clear idea about a MVP architecture around this, you are most welcome to write it and submit a pull request. I am rather busy with other things at the moment, but plan some minor changes in the following weeks.

kr15h, 2018-02-07:
The trick here would be to make a Selection vector that would hold references to the Vec objects. I think that would be a great first step. I will try to sketch this out as soon as possible.