kr15h/ofxPiMapper #106: build ofxpimapper on x86

pietrondo, 2017-05-05:
Hi.
I need to build ofxPiMapper on my x86 linux. How can I do?
thx (I try with make like on raspberry, but i have some errors)

kr15h, 2017-05-05:
try removing ofxOMXPlayer from addons.make

pietrondo, 2017-05-05:
```
`/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/lib/linux64/libopenFrameworks.a(ofImage.o): nella funzione "bool loadImage<unsigned char>(ofPixels_<unsigned char>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)":
ofImage.cpp:(.text+0x2460): riferimento non definito a "Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/lib/linux64/libopenFrameworks.a(ofImage.o): nella funzione "bool loadImage<unsigned short>(ofPixels_<unsigned short>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)":
ofImage.cpp:(.text+0x2be0): riferimento non definito a "Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/lib/linux64/libopenFrameworks.a(ofImage.o): nella funzione "bool loadImage<float>(ofPixels_<float>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)":
ofImage.cpp:(.text+0x30e0): riferimento non definito a "Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/lib/linux64/libopenFrameworks.a(ofUtils.o): nella funzione "ofLaunchBrowser(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)":
ofUtils.cpp:(.text+0x9ec9): riferimento non definito a "Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
ofUtils.cpp:(.text+0x9f07): riferimento non definito a "Poco::URI::setQuery(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
ofUtils.cpp:(.text+0x9f33): riferimento non definito a "Poco::URI::toString[abi:cxx11]() const"
ofUtils.cpp:(.text+0xa0a8): riferimento non definito a "Poco::URI::toString[abi:cxx11]() const"
/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/lib/linux64/libopenFrameworks.a(ofURLFileLoader.o): nella funzione "ofURLFileLoaderImpl::threadedFunction()":
ofURLFileLoader.cpp:(.text+0x54d2): riferimento non definito a "Poco::Thread::setName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
ofURLFileLoader.cpp:(.text+0x6d8a): riferimento non definito a "Poco::SystemException::SystemException(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)"
ofURLFileLoader.cpp:(.text+0x6f2b): riferimento non definito a "Poco::SystemException::SystemException(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)"
/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/lib/linux64/libopenFrameworks.a(ofURLFileLoader.o): nella funzione "ofURLFileLoaderImpl::handleRequest(ofHttpRequest)":
ofURLFileLoader.cpp:(.text+0x7e23): riferimento non definito a "Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
ofURLFileLoader.cpp:(.text+0x7e35): riferimento non definito a "Poco::URI::getPathAndQuery[abi:cxx11]() const"
ofURLFileLoader.cpp:(.text+0x7e5b): riferimento non definito a "Poco::Net::HTTPMessage::HTTP_1_1[abi:cxx11]"
ofURLFileLoader.cpp:(.text+0x7e65): riferimento non definito a "Poco::Net::HTTPRequest::HTTP_GET[abi:cxx11]"
ofURLFileLoader.cpp:(.text+0x7e75): riferimento non definito a "Poco::Net::HTTPRequest::HTTPRequest(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
ofURLFileLoader.cpp:(.text+0x7e9c): riferimento non definito a "Poco::Net::NameValueCollection::add(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
ofURLFileLoader.cpp:(.text+0x7f09): riferimento non definito a "Poco::Net::HTTPSClientSession::HTTPSClientSession(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short)"
ofURLFileLoader.cpp:(.text+0x8187): riferimento non definito a "Poco::Net::HTTPClientSession::HTTPClientSession(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short)"
ofURLFileLoader.cpp:(.text+0x82c7): riferimento non definito a "Poco::Exception::displayText[abi:cxx11]() const"
ofURLFileLoader.cpp:(.text+0x83cf): riferimento non definito a "Poco::Exception::displayText[abi:cxx11]() const"
/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/lib/linux64/libopenFrameworks.a(ofURLFileLoader.o): nella funzione "ofURLFileLoaderImpl::ofURLFileLoaderImpl()":
ofURLFileLoader.cpp:(.text+0x9693): riferimento non definito a "Poco::Net::Context::Context(Poco::Net::Context::Usage, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Poco::Net::Context::VerificationMode, int, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)"
ofURLFileLoader.cpp:(.text+0x9c1b): riferimento non definito a "Poco::Exception::displayText[abi:cxx11]() const"
[... cut ...]
X509Certificate.cpp:(.text+0xf71): riferimento non definito a "sk_value"
collect2: error: ld returned 1 exit status
make[1]: *** [/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:384: bin/example] Error 1
make[1]: uscita dalla directory "/home/pietro/Scaricati/of_v0.9.8_linux64_release/addons/ofxPiMapper/example"
make: *** [/home/pietro/Scaricati/of_v0.9.8_linux64_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:130: Release] Error 2

```

kr15h, 2017-05-05:
are you able to compile the example projects?

doctea, 2017-05-06:
Looks like POCO might not be installed properly?

I've been compiling and working with ofxPiMapper on Linux x64 so it is possible, I don't know if there are more steps required.  

@pietrondo are you really trying to build for x86 as the issue title says, or for x64 as the errors say?

pietrondo, 2017-05-06:
yes.. is a problem with archlinux (and poco) with all of Example

Pietro Capriata

http://www.pietrocapriata.me

CONFIDENTIALITY DISCLAIMER
Le informazioni contenute in questo messaggio di posta elettronica e negli
allegati sono riservate e confidenziali e ne sono vietate la lettura,
l'uso, la copia, la comunicazione e la diffusione in qualunque modo
eseguite. Qualora lei non fosse la persona destinataria del messaggio, la
invitiamo ad eliminarlo, dando gentilmente comunicazione al mittente
tramite e-mail di ritorno.

The information contained in this email message and any files transmitted
with it are confidential and privileged and any reading, processing,
distribution or copy of this material is strictly prohibited, in any form.
If you are not the intended recipient of this message, please immediately
delete it, giving the relevant communication to the sender by reply e-mail.

pietrondo, 2017-05-08:
ok on my girlfriend ubuntu it work.. but the video doesn't start (only sound)

pietrondo, 2017-06-06:
ok. with last git version ok on x86!
thx

closed 2017-06-06 by pietrondo

kr15h, 2017-06-06:
you probably had the old master branch which was supposed to work with openFrameworks git master branch not the stable version. Glad that you solved it.