kr15h/ofxPiMapper #113: Error when ofxPiMapper to a project

pauloguerraf, 2017-06-06:
Hi,

I am trying to use ofxPiMapper in a project, but when I add it as addon to a project I keep getting the following errors:

ofxpimapper\src\surfaces\gridwarpsurface.cpp(63): error C4716: 'ofx::piMapper::GridWarpSurface::setGridRows': must return a value
ofxpimapper\src\surfaces\gridwarpsurface.cpp(68): error C4716: 'ofx::piMapper::GridWarpSurface::setGridCols': must return a value
ofxpimapper\src\application\settingsloader.cpp(238): error C4716: 'ofx::piMapper::SettingsLoader::save': must return a value

Any help to solve will be greatly appreciated. 

Best,

Paulo

kr15h, 2017-06-06:
What is the platform you are building it on? And the version of openFrameworks?

pauloguerraf, 2017-06-06:
Hi, I am using openFrameworks 0.9.8. and building for Windows.

kr15h, 2017-06-06:
Could you post more of the error content you got?

pauloguerraf, 2017-06-06:
Here is a screenshot of what I get... I added ofxPiMapper and
ofxXmlSettings to the project addons, and in ofApp.h I added these 2 lines
#include "ofxXmlSettings.h"
#include "ofxPiMapper.h"

the project was working fine before that.

[image: Inline image 2]

pauloguerraf, 2017-06-06:
got it working

added return _gridRows in line 63
added return _gridCols in line 68
and added a boolean return depending on the result of xml.save in line 238

kr15h, 2017-06-06:
Good. Usually these are warnings with other compilers.

closed 2017-10-16 by kr15h