kr15h/ofxPiMapper #114: mapped mouse coordinates inside and fbo source

pauloguerraf, 2017-06-13:
Hello again :)

More than an issue, this is a question. Is there any way to get the mapped mouseX and mouseY coordinates on a mapped surface? If I draw an ofEllipse using ofGetMouseX() and ofGetMouseY() on an Custom FBO Source it does draw correctly on the mapped surface. Nevertheless if I try to change its color based on the mousePosition it uses the application absolute coordinates and not the coordinates I am using to draw within the FBO source. I am trying to get some interactivity working within each FBO source in my application. I would really appreciate any advice or tip to get this working.
Thank you in advance for your help.
Paulo

kr15h, 2017-06-14:
Hi! This is a nice idea to have those. Now there is not an option to do that, but I suppose that it could be cool to have something like the following.

```
ofPoint local = surface.globalToLocal(x, y);
ofPoint global = surface.localToGlobal(x, y);
```

Will put this on my list.

pauloguerraf, 2017-06-14:
Sounds great..  thank you!