kr15h/ofxPiMapper #115: absolute vs relative paths for sources

pauloguerraf, 2017-06-14:
Hi again,

One more issue I have found working on Windows. It seems ofxPiMapper saves the absolute path of files to the xml settings files when saving a composition (please correct me if I am wrong). This is causing a problem when trying to rerun the project. I am receiving the following error messages as if ofxPiMapper couldn't find the files listed in the xml settings. Maybe using relative instead of absolute paths could help?
Once again, thank you for all your help. Please let me know if there is anything I can help with. Would love to contribute to your project.

Best,

Paulo
[image]

kr15h, 2017-06-15:
These are not error messages. These are notifications (notices) about something happening internally. Is the application crashing?

kr15h, 2017-06-15:
You can also see how the paths are saved by opening "data/ofxpimapper.xml" file in any text editor. For now the paths are relative to the data/sources folder.

pauloguerraf, 2017-06-15:
Yes... the problem I see is exactly in how the paths are been saved. At least in Windows. The application is not crashing, but after saving a composition,  assigning an image or video  a to surface and closing the app, if I open ofxpimapper.xml in any text editor I see the path saved as an absolute path : C:/Users/pauloguerraf/Documents/ofv9_8_release/apps/myApps/piMapperTest/bin/data/sources/images/image1.png

 If try to reopen the application I get all those notices. The surface is there but the image isn't. When I read the console output I notice that ofxPiMapper Media Server is not finding the file as if it is trying to use the absolute path as a relative path. If I go back to the ofxPiMapper.xml and leave the path only as images/image1.png the application works again. The problem is that everytime I close the application the path gets rewritten as C:/Users/pauloguerraf/Documents/ofv9_8_release/apps/myApps/piMapperTest/bin/data/sources/images/image1.png

kr15h, 2017-06-15:
Ok, seems like this has to be fixed.

toughvj, 2017-09-20:
Any news on that? I have similar problem with medias from usb, app writes to xml source like /sources/videos/test.mp4 instead of /media/usb0/test.mp4.
Also when I define custom PI_VIDEOS_DIR in MediaServer.h it won't find any files.

kr15h, 2017-09-21:
Ok. I see. Will try to fix this.

dasoe, 2020-11-09:
Hi all,

**First of all: Thanks a lot for this incredible piece of software!**

I just stumbled upon this issue, while using ofxPiMpper for the first time on windows. 
I solved it the hard way for me by changing ofToDataPath() to relative.

MediaServer.cpp, Line 17/18:
`imageWatcher(ofToDataPath(DEFAULT_IMAGES_DIR, false), SourceType::SOURCE_TYPE_IMAGE),`
`videoWatcher(ofToDataPath(DEFAULT_VIDEOS_DIR, false), SourceType::SOURCE_TYPE_VIDEO)`

SettingsLoader.cpp, Line 84:
`// Construct full path`
`std::string dir = mediaServer.getDefaultMediaDir(typeEnum);`
`std::stringstream pathss;`
`pathss << ofToDataPath(dir,false) << sourceName;`
`std::string sourcePath = pathss.str();`

I'd love to do a Pull request, but the change will have consequences, so to make it right, I should probably first understand why you included absolute paths in the first place and why windows reacts differently. Any hint from somebody knowing the software would be helpful. Also: At the moment there's only TARGET_RASPBERRY_PI or not? No other target/system differentiations in code?

have a great day!
oe

kr15h, 2020-11-09:
Thanks!

I believe the absolute paths are there to make it work on the Raspberry Pi. Pull request would be great. 

—-
Krisjanis Rijnieks
Creative coding. Projection mapping. Solutions
https://rijnieks.com