kr15h/ofxPiMapper #117: Video Sync

mungle, 2017-07-26:
Hello,
is it possible to sync videos over network?
Thank you

kr15h, 2017-07-26:
One way would be to encode the video so you can seek to frames, upload the video to all your RasPi devices and add OSC receivers to receive sync messages from a main RasPi that is playing back the same video and sending frame numbers to all connected RasPi's at once.

Another way would be to stream video from a computer or Raspberry Pi and use omxPlayer to connect to that stream from all of your Raspberry Pi's.

magdesign, 2017-07-26:
Once tried to sync with this approach: https://github.com/turingmachine/omxplayer-sync
But the problem which must to be solved first: https://github.com/jvcleave/ofxOMXPlayer/issues/96

pietrondo, 2017-11-24:
same problem.
I need to sync two video on two raspberry (omxplayer-sync works very well), but i need also some mapping..
Other solution?

magdesign, 2017-11-24:
There is only a hardware solution so far.
I bought a HDMI recorder on ebay, doing the mapping with PiMapper, record the output, put it as videofile back onto the pi and then use omxplayer-sync.

Neon22, 2017-11-24:
does VLC allow this ? I know it streams well...

magdesign, 2017-11-25:
If we are talking about raspberrypi, there is only omxplayer available for the solution since no other player is supporting the hardware chip.

You could fork ofxomxplayer and add the missing dbus support, then I will make the syncscript.

pietrondo, 2017-11-25:
yep. at the moment we make an installation with omxplayer-sync . but the best is to use ofxpimapper.
but ofxomxplayer doesn't support the sync at the moment

magdesign, 2020-02-26:
By now it should be possible to sync several PiMapper RPis over the network since [ofxOMXPlayer](https://github.com/jvcleave/ofxOMXPlayer) was rewritten from scratch.
The `example-shader` has controls to pause and seek forward a video and also displays the current frame number.

What we now need is an option to start ofxPiMapper as slave, so it would start receiving the timecode from another PiMapper instance, check if it is in the same position, if not, seek forward, a bit ahead of the masters timecode, go to pause and when the master reached the same time, start the video. 

We made a [script](https://github.com/turingmachine/omxplayer-sync) with this approach for omxplayer and it works very well.

Any motivated openframeworks coder how can help with this please contact me for paid work.

kr15h, 2020-02-28:
Good to know that OMXPlayer got rewritten. I am trying to source enough time to take a closer look.

mmunderlinegithub, 2021-05-05:
Hey magdesign! (greetings form Berne ;-))
Did you manage to make it work? I'm looking exactly for the same thing. I got my Pi 3 B+ working with omxplayer-sync and ofxPimapper. Would super cool now to use the sync feature from OMXPlayer in the mighty ofPiMapper. What do you think?
Thanks a lot! Greetings, MM

magdesign, 2021-05-05:
@mmunderlinegithub 
yes, sync mapping is possible. 
I sponsored a developer to solve mapping sync over network,
its available with my outstanding [PocketVJ Exhibition](https://www.pocketvj.com/pocketvj-exhibition)

mmunderlinegithub, 2021-05-05:
@magdesign 
thank you for the fast reply. alright, cool. Unfortunatly I already bought my Pis and thus can't afford your soultion. :(.
Is there a way to buy just the script to link those two tools? thank you in any case

mmunderlinegithub, 2021-05-06:
@magdesign sorry again. If a buy two of these: https://www.pi-shop.ch/raspberry-pi-1-model-b. I'll be able to run your software (https://github.com/magdesign/PocketVJ) on it to sync a video while mapping it, right? I'll make a small donation if it works. :)