kr15h/ofxPiMapper #118: last openframeworks git

pietrondo, 2017-07-26:
Hi it is working ofxPimapper with last git of openframeworks?
I have archlinux and 0.9.8 have some problem with it.

thx

kr15h, 2017-07-26:
No, I decided to develop only against the current stable (0.9.8) version.

toughvj, 2017-09-20:
Hi there, so what is the best of version to build? I'm running Raspbian Stretch on RPI3.
regards!

kr15h, 2017-09-20:
Hi! The one hat the master branch is at currently.

toughvj, 2017-09-20:
Thank's for reply!
I mean openframeworks :) You also?
When using (0.9.8) during compilation a receive many notifications about none return statements and unused variables, ignore this or add return statements and remove variables?

kr15h, 2017-09-20:
Yes, 0.9.8 is the current stable version. You can ignore the warnings. Errors are what matter.

closed 2017-10-16 by kr15h