kr15h/ofxPiMapper #120: Slideshow of all images

magdesign, 2017-10-14:
Display a slideshow of all images in one folder.

How can ImageSource.cpp be modified to play all images in a folder as a sequence with e.g-3s delay?

Would be lovely, to make all images same size I would run imagemagick script before starting piMapper.
Or is the approach more like animating "select next source" keystroke with a keyboard emulator script?

Thanks for any hints.

kr15h, 2017-10-15:
Hi! I would recommend to create a source based on the FboSource class.

```
// ImageSlideshowSource.h

#pragma once

#include "ofMain.h"
#include "FboSource.h"

class ImageSlideshowSource : public ofx::piMapper::FboSource{
public:
    void setup(){
        name = "Image Slideshow"
        allocate(800, 600);
        // Load your images here
    }

    void draw(){
        // Draw your images here
    }
};
```

And then in ofApp

```
// ofApp.cpp

void ofApp::setup(){
    slideshow = new ImageSlideshowSource;
    mapper.registerFboSource(slideshow);
}

```

There is a bit more to it, but if you can't figure it out, I will edit this to a more complete version.

magdesign, 2017-11-06:
ok.
I engage a coder on this.
Made a fork and when its finished I will make a pull request.

kr15h, 2017-11-06:
Ok cool

magdesign, 2017-12-05:
I close this issue since Cris(tobal) Mendoza is now coding this feature and we are soon ready to commit.
https://github.com/magdesign/ofxPiMapper/tree/magSlideShowSource

kr15h, 2017-12-05:
Ok, let's see how it goes. Thanks!

kr15h, 2017-12-05:
Maybe it is a bit too early to close then. Let's close it when it is ready.

magdesign, 2017-12-08:
The slideshow function is now finished and working.
You can clone and test it from the master branch here: https://github.com/magdesign/ofxPiMapper

We also have a DirectoryWatcher which checks if a slide is added or removed.
Would be cool if you could merge the whole thing into your branch.
Let me know what you think.

kr15h, 2017-12-18:
Nice. Will check it out!

kr15h, 2017-12-18:
Can you submit a pull request?

kr15h, 2017-12-18:
No worries, got it compiling.

kr15h, 2017-12-18:
Merged with master branch. Thank you very much!

closed 2017-12-18 by kr15h

magdesign, 2017-12-18:
You are welcome.
That's the way opensource works.... :-)