kr15h/ofxPiMapper #121: new MediaServer

toughvj, 2017-10-24:
Hi, with your change of media server source code there is a problem, that when loading program it' tries to copy videos and images from data directory to data directory. Perhaps you should remove lines:

imageWatcher(ofToDataPath(DEFAULT_IMAGES_DIR, true), SourceType::SOURCE_TYPE_IMAGE),
	videoWatcher(ofToDataPath(DEFAULT_VIDEOS_DIR, true), SourceType::SOURCE_TYPE_VIDEO)
?

kr15h, 2017-10-25:
Thanks for noticing. That might cause slowdown.

toughvj, 2017-10-25:
I think this could help:

for(int i = 0; i < _directory.size(); ++i){
		_filePaths.push_back(path + _directory.getName(i));
		
		ofFile file(path + _directory.getName(i));
		if(_mediaType == SourceType::SOURCE_TYPE_VIDEO){
			if(!ofFile::doesFileExist("sources/videos/" + _directory.getName(i))){
				file.copyTo("sources/videos/" + _directory.getName(i));
				ofLogNotice("File: " + _directory.getName(i) + " added to source list");
			}
		}else if(_mediaType == SourceType::SOURCE_TYPE_IMAGE){
			if(!ofFile::doesFileExist("sources/images/" + _directory.getName(i))){
				file.copyTo("sources/images/" + _directory.getName(i));
				ofLogNotice("File: " + _directory.getName(i) + " added to source list");
			}
		}
	}

toughvj, 2017-10-31:
any news on this?

kr15h, 2017-10-31:
Not yet.