kr15h/ofxPiMapper #122: Outlines on png alpha image?

haelbichalex, 2017-10-30:
Hi!
So I am using a custom FBO source to render a background image and a PNG on top.
Without ofxPiMapper everything looks fine, but now I have these weird outlines around my png.
I tried using  `ofEnableAlphaBlending()` in both custom source and ofApp.cpp without success.
I also tried getting multisample antialiasing to work by using `fbo->allocate(width, height, GL_RGBA, 4)` in FboSource.cpp but then I only get a black screen.
Any hints?

[image]

kr15h, 2017-10-30:
Hi! How big is the image and can you send me the file?

haelbichalex, 2017-10-30:
It's 70KB small. It's a cutout of the [background](https://user-images.githubusercontent.com/15052352/32177697-37319318-bd8b-11e7-9110-8485810804e9.png) so it should blend perfectly in.
[image]

kr15h, 2017-10-30:
I see. Can I have also the background?

haelbichalex, 2017-10-30:
I just linked it in my previous answer 👍

kr15h, 2017-10-30:
got it. will take a look.

kr15h, 2017-10-30:
Which platform are you using? As on Mac it fits perfectly.

haelbichalex, 2017-10-30:
My destination platform is Windows 10 using Visual Studio 2015.
I just tried it on my Mac as well, but I get those outlines there as well 😕 
My custom source just looks like this, with '2.png' being the background:
```
#include "CustomSource.h"

void CustomSource::setup(){
	name = "Custom FBO Source";
	allocate(1920, 1080);

    img1.load("1.png");
    img2.load("2.png");
}

// Don't do any drawing here
void CustomSource::update(){
}

void CustomSource::draw(){
    img2.draw(0,0,1920,1080);
    img1.draw(0,0,1920,1080);
}
```

kr15h, 2017-10-30:
Ah you using custom source. Good. Now I have more hints.

kr15h, 2017-10-31:
This was due the ofFbo using different ways of blending. I fixed that by adding the following to the SurfaceStack in ofxPiMapper code.

```
glEnable(GL_BLEND);
glBlendFuncSeparate(GL_ONE, GL_ZERO, GL_ONE, GL_ONE_MINUS_SRC_ALPHA);
```

Check commit for more details.

haelbichalex, 2017-10-31:
So this did work on my Mac. Unfortunately on Windows the problem still exists.

kr15h, 2017-10-31:
OK. Let's see...

haelbichalex, 2017-11-02:
So does OF on Windows use another method for blending the fbo?
I tried updating to the latest version of [glew](https://github.com/nigels-com/glew).

kr15h, 2017-11-03:
I'm working on my windows virtualbox. unfortunatelly the internet where I am now is very slow and I am packed with deadlines. Soon...

commit b193764e (2017-10-31): Fix transp png images have border, close #122