kr15h/ofxPiMapper #123: example gamepad

magdesign, 2017-11-06:
What for is the example gamepad?
Does this work with a PS3 usb controller instead of the mouse?

kr15h, 2017-11-06:
Yes, that's the idea. It uses STL game engine.

magdesign, 2017-12-17:
This means its still necessary to start ofxPiMapper with an usb device connected?
I am trying to figure out if a GPIO gamepad like this https://www.ebay.com/itm/Keyestudio-RPI-JoyStick-Shield-DIY-Kits-Module-for-Simulated-Mouse/201831299666?hash=item2efe153a52:g:KtAAAOSwr~lYr6hp
would work as a mapping controller or if I still must connect a mouse or gamepad over usb.

magdesign, 2017-12-17:
I successfully compiled the example_gamepad after installing SDL:
sudo apt-get install libsdl2-dev
sudo apt-get install libsdl2-2.0-0

Then I plugged in an usb PS2 controller and started ofxPiMapper, all I got was outputs like:
SDL_JOYBUTTONDOWN
controllerIndex: 0
command: COMMAND_NONE

So would I have to map the controller commands to some actions or how would this work?
You mentioned something about STL game engine.

kr15h, 2017-12-18:
Yes, you have to map it. I started to develop an addon for this: https://github.com/kr15h/ofxJoystack There are a few things that have to be updated, but I could push the code in the following days for you to take a look

kr15h, 2017-12-18:
The sudo apt-get install's are installing the SDL game engine. It is needed in order to make the gamepads to work on the Raspberry Pi