kr15h/ofxPiMapper #128: Example compile error. Missing glm/geometric.hpp

Audiogeek1974, 2017-12-26:
Tried to download and compile the example project today and got the following error:

/home/pi/openFrameworks/addons/ofxPiMapper/src/Gui/Widgets/ScaleWidget.cpp:2:29: fatal error: glm/geometric.hpp: No such file or directory
 #include "glm/geometric.hpp"
                             ^
compilation terminated.
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxPiMapper/src/Gui/Widgets/ScaleWidget.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxPiMapper/src/Gui/Widgets/ScaleWidget.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxPiMapper/example'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

It appears to be missing geometric.hpp. Where is this file supposed to be and how can I get it?

Thanks!

kr15h, 2017-12-27:
Hi! It appears that is due to my latest changes to make it work with openFrameworks master branch. Which version of openFrameworks are you using and on which platform?

Audiogeek1974, 2017-12-27:
I thought it was the latest version 0.9.8 as I followed the instructions here:
http://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/

I'm running it on a Pi 2 with Raspbian Jessie Lite. I searched a little and found libglm-dev and installed it. That seemed to clear up the error. Maybe this just needs to be added as a dependency?

However, after compiling I still could not get the mapper example project to run. I tried to run OMXplayer, OMXplayer-sync, and HPlayer by themselves and had trouble with the sample videos included in the example project. 

I downloaded the bunny video from here:
wget https://github.com/jvcleave/ofxOMXPlayer/raw/master/video/Timecoded_Big_bunny_1.mov

And it played correctly. Is it possible these files have become corrupted? I can't paste in the errors here as I am currently starting a build on another SD card. This time I'm starting with full Jessie instead of Lite.

Audiogeek1974, 2017-12-27:
Tried downloading the current main branch of OF, but it appears to be quite a bit larger than the release. Had to get a bigger card and start over :)

kr15h, 2017-12-27:
It should work with oF 0.9.8 also. Let me check, it should not complain about the .hpp file.

kr15h, 2017-12-27:
There was a leftover include that was not supposed to be taken into account for the stable version. Improved it with 2a7416f663499c024f48db250f3b1f32a4ec955c Should work now.

Audiogeek1974, 2017-12-28:
On the bigger card, i was able to get the latest OF master branch and successfully build it. Tried compiling latest example project and got this new error:

/home/pi/openFrameworks/addons/ofxPiMapper/src/Application/SettingsLoader.cpp: In member function ‘bool ofx::piMapper::SettingsLoader::save(ofx::piMapper::SurfaceManager&, std::__cxx11::string)’:
/home/pi/openFrameworks/addons/ofxPiMapper/src/Application/SettingsLoader.cpp:246:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
cc1plus: some warnings being treated as errors
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:358: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxPiMapper/src/Application/SettingsLoader.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxPiMapper/src/Application/SettingsLoader.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxPiMapper/example'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2

This function at line 159 appears to be missing a return statement.
// TODO: Save all presets, not just the active one.
bool SettingsLoader::save(SurfaceManager & surfaceManager, std::string fileName){

Audiogeek1974, 2017-12-28:
I found a couple more. I'm just editing them with return true statements.

/home/pi/openFrameworks/addons/ofxPiMapper/src/Surfaces/GridWarpSurface.cpp: In member function ‘int ofx::piMapper::GridWarpSurface::setGridRows(int)’:
/home/pi/openFrameworks/addons/ofxPiMapper/src/Surfaces/GridWarpSurface.cpp:58:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
/home/pi/openFrameworks/addons/ofxPiMapper/src/Surfaces/GridWarpSurface.cpp: In member function ‘int ofx::piMapper::GridWarpSurface::setGridCols(int)’:
/home/pi/openFrameworks/addons/ofxPiMapper/src/Surfaces/GridWarpSurface.cpp:63:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^

Audiogeek1974, 2017-12-28:
Sorry for the dumb comment above. It would be a little difficult to return true for an int function :) I'm still new to all of this. Since these are "set" functions, should they just be voids?

Audiogeek1974, 2017-12-28:
Sorry, but I still can't seem to get this to compile and I don't really understand what this error is telling me. Any ideas? I feel like I must be missing something simple.

/usr/bin/ld: /home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofxOMXPlayer.o: undefined reference to symbol 'eglCreateImageKHR'
//opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:403: recipe for target 'bin/example' failed
make[1]: *** [bin/example] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxPiMapper/example'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2

kr15h, 2017-12-28:
Seems like you have to use the stable branch of the ofxOMXPlayer. Did you try the following? 
```
cd openFrameworks/addons/ofxOMXPlayer
git checkout 0.9.0-compatible
```

Audiogeek1974, 2017-12-29:
The nerve of picky compilers :)

Those are the exact two lines I used. Do I then need to build all of those examples?

I did notice that there was another omxplayer installed. I removed it. How is this different than ofxOMXPlayer?

Are you using the latest build of OF from the master branch or the nightly builds?

kr15h, 2017-12-31:
Fixed the non-void functions returning void in ef6ced67013c19bd3b827e22232bd2500447dd4c and 8cf585fb7b3a51756cfdc7ef351f97f2d928351a. ofxOMXPlayer, i suppose, uses the same libraries omxplayer is also built on, which would mean that, if you install a new version of omxplayer, ofxOMXPlayer gets updated as well.

closed 2017-12-31 by kr15h

Audiogeek1974, 2018-01-02:
Thank you. I think I have it all working now. I have a couple of remaining questions though.
What are the 3 selections under FBO supposed to do in the example? One shows an oscillating X with yellow background, the second shows black & white horizontal lines that appear to be rather random, and the one marked "slideshow" doesn't display anything. And what is this intended to be used for?

kr15h, 2018-01-02:
Hi! The slideshow is supposed to be a slideshow source that loads all the source images and loops them. I see that it has been broken after the last updates. Will fix it ASAP.