kr15h/ofxPiMapper #129: with last github opensourceframeworks: fatal error: ofxOMXPlayer.h: No such file or directory

pietrondo, 2018-01-08:
Hi!
I try to compile with the last stretch raspbian and last opensourceframeworks from github and..:

> Compiling example for Release
make[1]: Entering directory '/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libcurl glfw3 rtaudio gtk+-3.0 libmpg123 
with PKG_CONFIG_LIBDIR=
Compiling /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example/src/main.cpp
g++ -c -O3 -DNDEBUG -Wall -Werror=return-type -std=c++14 -DGCC_HAS_REGEX -march=armv6 -mfpu=vfp -mfloat-abi=hard -fPIC -ftree-vectorize -Wno-psabi -pipe -DOF_USING_GTK -DOF_USING_GTK -DOF_USING_MPG123 -DTARGET_RASPBERRY_PI -DSTANDALONE -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM  -I/opt/vc/include -I/opt/vc/include/IL -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -pthread -D_REENTRANT -pthread -I/usr/include/gstreamer-1.0 -I/usr/include/AL -I/usr/include/arm-linux-gnueabihf -I/usr/include/rtaudio -I/usr/include/alsa -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/glm/include -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/glm/include/glm -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/glm/include/glm/detail -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/glm/include/glm/simd -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/glm/include/glm/gtc -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/glm/include/glm/gtx -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/json/include -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/kiss/include -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/tess2/include -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/utf8/include -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/utf8/include/utf8 -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/graphics -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/math -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/events -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/video -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/communication -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/app -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/utils -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/3d -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/types -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/sound -I/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworks/gl -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example/src -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example/example.xcodeproj/xcshareddata -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example/example.xcodeproj/xcshareddata/xcschemes -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxGui/src -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Utils -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/UserInterface -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Sources -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Types -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/MediaServer -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Surfaces -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Info -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Commands -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Application -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Application/Modes -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Gui -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Gui/Widgets -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxXmlSettings/src -I/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxXmlSettings/libs -I/. -I. -I./obj -I./obj/linuxarmv6l -I./obj/linuxarmv6l/Release -I./obj/linuxarmv6l/Release/src -I/./bin -I./bin -I./bin/data -I./bin/data/sources -I./bin/data/sources/images -I./bin/data/sources/videos -I./src -I./example.xcodeproj -I./example.xcodeproj/xcshareddata -I./example.xcodeproj/xcshareddata/xcschemes -MMD -MP -MF obj/linuxarmv6l/Release/src/main.d -MT obj/linuxarmv6l/Release/src/main.o -o obj/linuxarmv6l/Release/src/main.o -c /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example/src/main.cpp
In file included from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/MediaServer/MediaServer.h:8:0,
                 from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Surfaces/SurfaceManager.h:7,
                 from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Gui/Widgets/ScaleWidget.h:8,
                 from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Gui/Gui.h:7,
                 from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Application/Application.h:13,
                 from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/ofxPiMapper.h:6,
                 from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example/src/ofApp.h:5,
                 from /home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example/src/main.cpp:2:
/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/src/Sources/VideoSource.h:7:27: fatal error: ofxOMXPlayer.h: No such file or directory
  #include "ofxOMXPlayer.h"
                           ^
compilation terminated.
/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:221: recipe for target 'obj/linuxarmv6l/Release/src/main.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/main.o] Error 1
make[1]: Leaving directory '/home/pietro/of_v20170714_linuxarmv6l_release/addons/ofxPiMapper/example'
/home/pietro/of_v20170714_linuxarmv6l_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2

kr15h, 2018-01-08:
Yes, that's cause I removed the `ofxOMXPlayer` entry from `addons.make`. Add it there, it should compile. I am thinking how it would be best to have it on RPi and not on other platforms.

pietrondo, 2018-01-09:
ok.. and now appear this problem:

> /usr/bin/ld: /home/pietro/of_v20170714_linuxarmv6l_release/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofxOMXPlayer.o: undefined reference to symbol 'eglCreateImageKHR'
//opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

pietrondo, 2018-01-09:
on ofxOMXplayer I have already: 
`git checkout 0.9.0-compatible`

kr15h, 2018-01-09:
That's interesting. I will have to try to reproduce this myself.

kr15h, 2018-01-10:
I added ofxOMXPlayer back to addons.make. Trying to reproduce the other error you have.

pietrondo, 2018-01-11:
Same errors..

> /usr/bin/ld: /home/pietro/of_v20170714_linuxarmv6l_release/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofxOMXPlayer.o: undefined reference to symbol 'eglCreateImageKHR'
//opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

I have already try to remove and reinstall ofxpimapper and ofxOMXplayer

pietrondo, 2018-01-12:
same error: latest github openframeworks and ofxPimapper

`/usr/bin/ld: /home/pietro/openFrameworks/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofxOMXPlayer.o: undefined reference to symbol 'eglCreateImageKHR'
//opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line
`

kr15h, 2018-01-12:
I think that there are changes in the Stretch version of Raspbian that have to be taken into account. I think one has to install latest Jessie and continue from there.

kr15h, 2018-01-12:
Also, just added instructions here: http://ofxpimapper.com/pages/github-instructions/
And it works like this. Another thing is that I have to add a part about openFrameworks EGLWindow patching to enable listening to all connected mice and keyboards.