kr15h/ofxPiMapper #131: change sources from command line?

ryanh31, 2018-01-26:
Hi
I am trying to sync the beginning and end of a source video file with the opening and closing of a projector shutter. I have a working python script that opens and closes a servo motor shutter. I want to start a movie in piMapper and open the servo at the same time. Right now it just opens and closes the shutter and the movie is playing on a loop using the example ofxPimapper file.

is there a way to call on the script with the command line and not the gui?

I think my options are calling the PiMapper script to start when the servo opens, or just change the source movie to a black image and when the servo is triggered, I then change the layer to my movie file.

When the servo shutter is closed, I'd like there to be a black screen. Ideally, the movie starts when the shutter opens.

is this possible?

kr15h, 2018-01-26:
Hi! Currently there is no command line feature like the one you are looking for, but it seems like a useful one. I would add a OSC layer in the ofxPiMapper application itself and send OSC messages from your Python script whenever you send the "open shutter" command. Do you have experience with openFrameworks?

ryanh31, 2018-01-26:
I don't have experience with open frameworks, but I do with Python and osc.

kr15h, 2018-01-26:
So, yes, if you try install ofxPIMapper from source, you should be able to edit the source files and add a bit of C++ code that handles the OSC part on the ofxPiMapper side. I plan to add an TCP and OSC interface for doing things like you want to do, but it will still take some time.

ryanh31, 2018-01-26:
I will give it a try. Thanks for your fast response.

CyrCom, 2018-03-16:
Hey @kr15h,
I am currently working on a TCP implementation of your script, it took me a lot of work to get there because I learned c++ and OF for that purpose, but I'm now familiar with TCP between Pure Data (Purr-Data) and OF. Pd as a server is temporary but easier for testing.
I will work this week on which functions to call in ofxPiMapper to achieve playing/pausing, source change, et caetera.
I work for theaters and this will be a terrific tool, because this is a cheap and elegant solution.
For instance, one of my projects uses 10+ video projections and everything needs to be controlled from far distance and be totally invisible. Also accessing physically the RPis is not always possible so adjusting the mapping by TCP would be great, if not mandatory.
This will allow cues and mapping to get mastered from a booth or with wifi from all over a place.
TCP is so fast that the Rpis will be pretty much synced while they have to play files.

Anyway, thanks for your great program, it is by far the cleanest and I'm very excited about those functionalities that will be (let's say it) revolutionary for my field of work.

This is more of a statement than a question actually :) but I see that you're still active on github so I'm thinking we could be in touch for this to happen.

There are actually things that I could ask:
Actually the only missing feature is to be able to play 1080p files. I tested and it's laggy but I'm sure we can get there, what do you think about this ?
Overclocking the Rpi or using the latest 3 B+ version for a 1.4GHz version.
Cutting out features that requires processing (maybe the keyboard input listening ?, ...)
ofOMXplayer adjustements (I really don't know), also why this and not Gstreamer or else ?
Are you going to port ofxPiMapper with OF 0.10.0 ? I'm not able to compile it with the nightly builds but I think it could use the improvements.

Compiling and improving are a little bit complicated for me because I'm new but maybe you would be interested there ;)